wdoekes/asterisk-chan-dongle #3: error building on debian jessie asterisk 11.13.1

ftoledo, 2015-10-17:
root@scarlet:~/src/wdoekes/asterisk-chan-dongle# aclocal
aclocal: warning: autoconf input should be named 'configure.ac', not 'configure.in'

root@scarlet:~/src/wdoekes/asterisk-chan-dongle# autoconf

root@scarlet:~/src/wdoekes/asterisk-chan-dongle# automake -a
automake: warning: autoconf input should be named 'configure.ac', not 'configure.in'
automake: error: no 'Makefile.am' found for any configure output

root@scarlet:~/src/wdoekes/asterisk-chan-dongle# ./configure 
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking target system type... x86_64-unknown-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
configure: error: Please set --with-astversion=MMmmuu (major.minor.micro)

root@scarlet:~/src/wdoekes/asterisk-chan-dongle# ./configure --with-astversion=11131
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking target system type... x86_64-unknown-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... none
checking how to run the C preprocessor... gcc -E
checking for strip... strip
checking for rm... rm
checking for library containing iconv... none required
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking fcntl.h usability... yes
checking fcntl.h presence... yes
checking for fcntl.h... yes
checking for stdlib.h... (cached) yes
checking for string.h... (cached) yes
checking sys/time.h usability... yes
checking sys/time.h presence... yes
checking for sys/time.h... yes
checking termios.h usability... yes
checking termios.h presence... yes
checking for termios.h... yes
[... 231 lines cut ...]
/usr/include/asterisk/channel.h:3757:13: note: previous declaration of ‘ast_channel_linkedid’ was here
 const char *ast_channel_linkedid(const struct ast_channel *chan);
             ^
In file included from app.c:22:0:
ast_compat.h: In function ‘ast_channel_linkedid’:
ast_compat.h:42:93: error: dereferencing pointer to incomplete type
 static inline const char *ast_channel_linkedid(const struct ast_channel *chan) { return chan->linkedid; }
                                                                                             ^
ast_compat.h: At top level:
ast_compat.h:45:20: error: static declaration of ‘ast_channel_hangupcause_set’ follows non-static declaration
 static inline void ast_channel_hangupcause_set(struct ast_channel *chan, int value) { chan->hangupcause = value; }
                    ^
In file included from /usr/include/asterisk/pbx.h:26:0,
                 from app.c:19:
/usr/include/asterisk/channel.h:3791:6: note: previous declaration of ‘ast_channel_hangupcause_set’ was here
 void ast_channel_hangupcause_set(struct ast_channel *chan, int value);
      ^
In file included from app.c:22:0:
ast_compat.h: In function ‘ast_channel_hangupcause_set’:
ast_compat.h:45:91: error: dereferencing pointer to incomplete type
 static inline void ast_channel_hangupcause_set(struct ast_channel *chan, int value) { chan->hangupcause = value; }
                                                                                           ^
Makefile:65: recipe for target 'app.o' failed
make: *** [app.o] Error 1
root@scarlet:~/src/wdoekes/asterisk-chan-dongle#

wdoekes, 2015-10-17:
MMmmuu = 111301

wdoekes, 2015-10-19:
From your silence, I conclude that my hint was sufficient. (Sorry for the brevity, it was typed on a smartphone.)

You're welcome. Closing out.

closed 2015-10-19 by wdoekes