wdoekes/asterisk-chan-dongle #5: Cannot build for asterisk 13.1.0

hzsolt94, 2016-05-10:
Everything seems fine until make command, which gives me a very long error list of "multiple definition".
`make
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT app.o -MF .app.o.d -MP  -o app.o -c app.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT at_command.o -MF .at_command.o.d -MP  -o at_command.o -c at_command.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT at_parse.o -MF .at_parse.o.d -MP  -o at_parse.o -c at_parse.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT at_queue.o -MF .at_queue.o.d -MP  -o at_queue.o -c at_queue.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT at_read.o -MF .at_read.o.d -MP  -o at_read.o -c at_read.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT at_response.o -MF .at_response.o.d -MP  -o at_response.o -c at_response.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT chan_dongle.o -MF .chan_dongle.o.d -MP  -o chan_dongle.o -c chan_dongle.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT channel.o -MF .channel.o.d -MP  -o channel.o -c channel.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT char_conv.o -MF .char_conv.o.d -MP  -o char_conv.o -c char_conv.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT cli.o -MF .cli.o.d -MP  -o cli.o -c cli.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT helpers.o -MF .helpers.o.d -MP  -o helpers.o -c helpers.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT manager.o -MF .manager.o.d -MP  -o manager.o -c manager.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT memmem.o -MF .memmem.o.d -MP  -o memmem.o -c memmem.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT ringbuffer.o -MF .ringbuffer.o.d -MP  -o ringbuffer.o -c ringbuffer.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT cpvt.o -MF .cpvt.o.d -MP  -o cpvt.o -c cpvt.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT dc_config.o -MF .dc_config.o.d -MP  -o dc_config.o -c dc_config.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT pdu.o -MF .pdu.o.d -MP  -o pdu.o -c pdu.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT mixbuffer.o -MF .mixbuffer.o.d -MP  -o mixbuffer.o -c mixbuffer.c
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT pdiscovery.o -MF .pdiscovery.o.d -MP  -o pdiscovery.o -c pdiscovery.c
gcc  -shared -Xlinker -x -o chan_dongle.so app.o at_command.o at_parse.o at_queue.o at_read.o at_response.o chan_dongle.o channel.o char_conv.o cli.o helpers.o manager.o memmem.o ringbuffer.o cpvt.o dc_config.o pdu.o mixbuffer.o pdiscovery.o 
at_command.o: In function `ast_atomic_fetchadd_int':
/usr/include/asterisk/lock.h:718: multiple definition of `ast_atomic_fetchadd_int'
app.o:/usr/include/asterisk/lock.h:718: first defined here
at_command.o: In function `ast_atomic_dec_and_test':
/usr/include/asterisk/lock.h:764: multiple definition of `ast_atomic_dec_and_test'
app.o:/usr/include/asterisk/lock.h:764: first defined here
at_command.o: In function `ast_tvdiff_sec':
/usr/include/asterisk/time.h:45: multiple definition of `ast_tvdiff_sec'
app.o:/usr/include/asterisk/time.h:45: first defined here
at_command.o: In function `ast_tvdiff_us':
/usr/include/asterisk/time.h:64: multiple definition of `ast_tvdiff_us'
app.o:/usr/include/asterisk/time.h:64: first defined here
at_command.o: In function `ast_tvdiff_ms':
/usr/include/asterisk/time.h:78: multiple definition of `ast_tvdiff_ms'
app.o:/usr/include/asterisk/time.h:78: first defined here
at_command.o: In function `ast_tvzero':
/usr/include/asterisk/time.h:95: multiple definition of `ast_tvzero'
app.o:/usr/include/asterisk/time.h:95: first defined here
at_command.o: In function `ast_tvcmp':
/usr/include/asterisk/time.h:106: multiple definition of `ast_tvcmp'
app.o:/usr/include/asterisk/time.h:106: first defined here
at_command.o: In function `ast_tveq':
/usr/include/asterisk/time.h:125: multiple definition of `ast_tveq'
app.o:/usr/include/asterisk/time.h:125: first defined here
at_command.o: In function `ast_tvnow':
/usr/include/asterisk/time.h:135: multiple definition of `ast_tvnow'
app.o:/usr/include/asterisk/time.h:135: first defined here
at_command.o: In function `ast_tv':
/usr/include/asterisk/time.h:182: multiple definition of `ast_tv'
app.o:/usr/include/asterisk/time.h:182: first defined here
at_command.o: In function `ast_samp2tv':
/usr/include/asterisk/time.h:197: multiple definition of `ast_samp2tv'
app.o:/usr/include/asterisk/time.h:197: first defined here
at_command.o: In function `_ast_malloc':
/usr/include/asterisk/utils.h:535: multiple definition of `_ast_malloc'
app.o:/usr/include/asterisk/utils.h:535: first defined here
at_command.o: In function `_ast_calloc':
/usr/include/asterisk/utils.h:559: multiple definition of `_ast_calloc'
app.o:/usr/include/asterisk/utils.h:559: first defined here
at_command.o: In function `_ast_realloc':
/usr/include/asterisk/utils.h:596: multiple definition of `_ast_realloc'
app.o:/usr/include/asterisk/utils.h:596: first defined here
at_command.o: In function `_ast_strdup':
/usr/include/asterisk/utils.h:624: multiple definition of `_ast_strdup'
app.o:/usr/include/asterisk/utils.h:624: first defined here
at_command.o: In function `_ast_strndup':
/usr/include/asterisk/utils.h:654: multiple definition of `_ast_strndup'
app.o:/usr/include/asterisk/utils.h:654: first defined here
[... 1952 lines cut ...]
app.o:/usr/include/asterisk/strings.h:992: first defined here
pdiscovery.o: In function `ast_str_set_escapecommas':
/usr/include/asterisk/strings.h:999: multiple definition of `ast_str_set_escapecommas'
app.o:/usr/include/asterisk/strings.h:999: first defined here
pdiscovery.o: In function `ast_str_append_escapecommas':
/usr/include/asterisk/strings.h:1006: multiple definition of `ast_str_append_escapecommas'
app.o:/usr/include/asterisk/strings.h:1006: first defined here
pdiscovery.o: In function `ast_str_set':
/usr/include/asterisk/strings.h:1034: multiple definition of `ast_str_set'
app.o:/usr/include/asterisk/strings.h:1034: first defined here
pdiscovery.o: In function `ast_str_append':
/usr/include/asterisk/strings.h:1060: multiple definition of `ast_str_append'
app.o:/usr/include/asterisk/strings.h:1060: first defined here
pdiscovery.o: In function `ast_check_digits':
/usr/include/asterisk/strings.h:1081: multiple definition of `ast_check_digits'
app.o:/usr/include/asterisk/strings.h:1081: first defined here
pdiscovery.o: In function `ast_tech_to_upper':
/usr/include/asterisk/strings.h:1099: multiple definition of `ast_tech_to_upper'
app.o:/usr/include/asterisk/strings.h:1099: first defined here
pdiscovery.o: In function `ast_rq_is_int':
/usr/include/asterisk/config.h:1116: multiple definition of `ast_rq_is_int'
app.o:/usr/include/asterisk/config.h:1116: first defined here
collect2: error: ld returned 1 exit status
Makefile:57: recipe for target 'chan_dongle.so' failed
make: *** [chan_dongle.so] Error 1`

wdoekes, 2016-05-10:
Next time, the steps preceding the make would've been nice.

And the gcc version.

But I think I can add two and two together and conclude that:
(a) you run Debian or a derivative with a very old Asterisk 13.1
(b) you run quite a new gcc 5 compiler

That means you lack this changeset:
https://github.com/asterisk/asterisk/commit/e48f2e7897079936ef2cb91fd15b88c319a3cdc9#diff-47b4c4af4bc4e5e2b56157c25afbbe86L51

This should do the trick for you:
```
--- /usr/include/asterisk/inline_api.h.orig	2016-05-10 22:25:56.437154055 +0200
+++ /usr/include/asterisk/inline_api.h	2016-05-10 22:26:00.829154734 +0200
@@ -46,7 +46,7 @@
 #if !defined(LOW_MEMORY) && !defined(DISABLE_INLINE)
 
 #if !defined(AST_API_MODULE)
-#define AST_INLINE_API(hdr, body) hdr; extern inline hdr body
+#define AST_INLINE_API(hdr, body) static hdr; static inline hdr body
 #else
 #define AST_INLINE_API(hdr, body) hdr; hdr body
 #endif
```

hzsolt94, 2016-05-12:
I'm very sorry, I wanted to attach the additional information.
The affected system is Ubuntu 16.04 LTS, so I think the problem will persist for a longer time.

Preceeding steps was:
``git clone https://github.com/wdoekes/asterisk-chan-dongle.git
cd asterisk-chan-dongle/
aclocal
autoconf
automake -a
./configure --with-astversion=130100
make``

GCC version:  gcc (Ubuntu 5.3.1-14ubuntu2) 5.3.1 20160413
Asterisk version: Asterisk 13.1.0~dfsg-1.1ubuntu4
Uname: Linux test 4.4.0-22-generic #39-Ubuntu SMP Thu May 5 16:53:32 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

Thank you very much for the patch, it solves the compilation problem! However I did not recompiled asterisk which your patch applies to, so I don't know if the binary will work or not. I'll test it, and provide feedback.

hzsolt94, 2016-05-12:
The chan_dongle compiled with the patch seems working OK. There are some instability when the dongle is "reused": Dongle needs reset before every asterisk start. I assume it's not related.

Thank you very much for your excellent support, feel free to let me know if I can do any related work in return.

closed 2016-05-12 by hzsolt94

wdoekes, 2016-05-14:
Glad it works. And no, you did not / should not have to recompile asterisk.

As for the reset before reuse, that should very much be unrelated indeed. You'll need to come up with a lot more debug output to get that fixed ;-)