wdoekes/asterisk-chan-dongle #8: Reception problem for certain sms

pjmichel, 2016-09-19:
Hi, i use a E180 Huawei, and have this kind of problem 

if i send a sms to the dongle like : essai 

no prob, but 
essai with a smiley or a accent,

[Sep 19 15:06:20] ERROR[17076]: at_response.c:1259 at_response_cmgr: [dongle1] Error decode SMS text 'D83DDE03D83DDE0E0020' from encoding 2, message is '+CMGR: 0,,29
07913306092041F0040B913306991364F60008619091516061800AD83DDE03D83DDE0E0020'

Any idea ?

fadasi, 2016-09-29:
Use https://www.diafaan.com/sms-tutorials/gsm-modem-tutorial/online-sms-pdu-decoder/ to analyze the response -> "Data coding: Unicode".

In char_conv.c replace "UCS-2BE" with "UTF-16BE", iit seems to work for all sms.

pjmichel, 2016-09-30:
On Thu, 29 Sep 2016 14:52:36 -0700
fadasi <notifications@github.com> wrote:

> Use
> https://www.diafaan.com/sms-tutorials/gsm-modem-tutorial/online-sms-pdu-decoder/
> to analyze the response -> "Data coding: Unicode".
> 
> In char_conv.c replace "UCS-2BE" with "UTF-16BE", iit seems to work
> for all sms.
> 

Very good, that solve the problem, i change ucs-2BE in 2 function ( one
from utf-8 one to utf8 ) perfect .

wdoekes, 2016-09-30:
```
>>> print b16decode('07913306092041F0040B913306991364F60008619091516061800AD83DDE03D83DDE0E0020'[22:]).decode('utf-16be')
㌆餓 憐酑恡耊😃😎 
```

commit bf38e687 (2016-09-30): Fix decoding of SMS with non-language characters.

Character conversion assumed UCS-2, which handles only the Basic
Multilingual Plane, but various parties will have switched to using
UTF-16 there.

Fix suggested by @fadasi. Reported and tested by @pjmichel. Thanks!
Closes #8.
closed 2016-09-30 by wdoekes