wdoekes/asterisk-chan-dongle #12: Add DongleSendUSSD in extension.conf

fadasi, 2016-10-01:
I need to use a DongleSendUSSD function in extensions.conf :
```
root@pabx:~# grep DongleSendUSSD /etc/asterisk/extensions.conf
exten => *#123#,n,DongleSendUSSD(dongle1,${EXTEN})
root@pabx:~#
```
to check the balance of a SIM (after a "Dial(Dongle/...", etc).

I found a code example at http://a.dmin.pro/?p=2712 (thanks Google) and I modified sources app.c:
[asterisk-chan-dongle-fadasi-3.zip](https://github.com/wdoekes/asterisk-chan-dongle/files/504533/asterisk-chan-dongle-fadasi-3.zip)

It would be nice to add this feature.

wdoekes, 2016-10-02:
It helps if you try and make a pull request of your changes instead of zipping a diff.

It's real easy once you get the hang of it. Reviewing those changes can then be done from the github interface.

fadasi, 2016-10-02:
Done.
Sorry for inconvenience, but I did not know how to do it.

closed 2016-10-05 by fadasi