wdoekes/asterisk-chan-dongle #15: Something wrong since you last commit

fadasi, 2016-10-02:
Yesterday when I compiled chan_dongle to try https://github.com/wdoekes/asterisk-chan-dongle/pull/10 (with a new "git clone https://github.com/wdoekes/asterisk-chan-dongle.git"), everything worked fine.

I am using (for some time) a Debian Jessie (last Raspbian) + asterisk (1:11.13.1~dfsg-2+b1) + asterisk-dev (1:11.13.1~dfsg-2).

Something wrong now, the compilation process executes correctly but chan_dongle is no longer seen by Asterisk since your latest commit (1e60445 4 hours ago ?):
```
root@pabx:~# cd github
root@pabx:~/github# cd wdoekes
root@pabx:~/github/wdoekes# rm -rf asterisk-chan-dongle
root@pabx:~/github/wdoekes# git clone https://github.com/wdoekes/asterisk-chan-dongle.git
Clonage dans 'asterisk-chan-dongle'...
remote: Counting objects: 445, done.
remote: Compressing objects: 100% (63/63), done.
remote: Total 445 (delta 21), reused 0 (delta 0), pack-reused 382
Réception d'objets: 100% (445/445), 302.71 KiB | 0 bytes/s, fait.
Résolution des deltas: 100% (240/240), fait.
Vérification de la connectivité... fait.
root@pabx:~/github/wdoekes# cd asterisk-chan-dongle
root@pabx:~/github/wdoekes/asterisk-chan-dongle# aclocal
aclocal: warning: autoconf input should be named 'configure.ac', not 'configure.in'
root@pabx:~/github/wdoekes/asterisk-chan-dongle# autoconf
root@pabx:~/github/wdoekes/asterisk-chan-dongle# automake -a
automake: warning: autoconf input should be named 'configure.ac', not 'configure.in'
configure.in:61: installing './compile'
configure.in:6: installing './config.guess'
configure.in:6: installing './config.sub'
configure.in:7: installing './install-sh'
configure.in:7: installing './missing'
automake: error: no 'Makefile.am' found for any configure output
root@pabx:~/github/wdoekes/asterisk-chan-dongle# ./configure --with-astversion=111301
checking build system type... armv7l-unknown-linux-gnueabihf
checking host system type... armv7l-unknown-linux-gnueabihf
checking target system type... armv7l-unknown-linux-gnueabihf
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking for style of include used by make... GNU
checking dependency style of gcc... none
checking how to run the C preprocessor... gcc -E
checking for strip... strip
checking for rm... rm
checking for library containing iconv... none required
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking fcntl.h usability... yes
checking fcntl.h presence... yes
[... 61 lines cut ...]
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT memmem.o -MF .memmem.o.d -MP  -o memmem.o -c memmem.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT ringbuffer.o -MF .ringbuffer.o.d -MP  -o ringbuffer.o -c ringbuffer.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT cpvt.o -MF .cpvt.o.d -MP  -o cpvt.o -c cpvt.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT dc_config.o -MF .dc_config.o.d -MP  -o dc_config.o -c dc_config.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT pdu.o -MF .pdu.o.d -MP  -o pdu.o -c pdu.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT mixbuffer.o -MF .mixbuffer.o.d -MP  -o mixbuffer.o -c mixbuffer.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT pdiscovery.o -MF .pdiscovery.o.d -MP  -o pdiscovery.o -c pdiscovery.c
gcc  -shared -Xlinker -x -o chan_dongle.so app.o at_command.o at_parse.o at_queue.o at_read.o at_response.o chan_dongle.o channel.o char_conv.o cli.o helpers.o manager.o memmem.o ringbuffer.o cpvt.o dc_config.o pdu.o mixbuffer.o pdiscovery.o
root@pabx:~/github/wdoekes/asterisk-chan-dongle# make install
strip chan_dongle.so
/usr/bin/install -c -m 755 chan_dongle.so /usr/lib/asterisk/modules
root@pabx:~/github/wdoekes/asterisk-chan-dongle# service asterisk restart
root@pabx:~/github/wdoekes/asterisk-chan-dongle# asterisk -r
Asterisk 11.13.1~dfsg-2+b1, Copyright (C) 1999 - 2013 Digium, Inc. and others.
Created by Mark Spencer <markster@digium.com>
Asterisk comes with ABSOLUTELY NO WARRANTY; type 'core show warranty' for details.
This is free software, with components licensed under the GNU General Public
License version 2 and other licenses; you are welcome to redistribute it under
certain conditions. Type 'core show license' for details.
=========================================================================
Connected to Asterisk 11.13.1~dfsg-2+b1 currently running on raspberrypi (pid = 21315)
pabx*CLI> dongle show devices
No such command 'dongle show devices' (type 'core show help dongle show' for other possible commands)
pabx*CLI>
```

wdoekes, 2016-10-02:
Interesting. I have indeed broken something there.

TBH, this is the first time I've actually loaded up chan_dongle into Asterisk. (I don't own a dongle.) And I can confirm that c24a3150a breaks things. I'll go do some digging to find out what I broke.

fadasi, 2016-10-02:
git checkout c1109d6 # Ok
git checkout c24a315 # NOk

Test procedure:
```
rm -rf asterisk-chan-dongle
git clone https://github.com/wdoekes/asterisk-chan-dongle.git
cd asterisk-chan-dongle
git checkout c1109d6
aclocal
autoconf
automake -a
./configure --with-astversion=111301
make
make install
cd ..
service asterisk restart
asterisk -rx "dongle show devices"
```

commit 428a746e (2016-10-02): Unbreak compiled module loading broken in c24a3150.

Fixes #15 but doesn't explain it yet.
commit dcf4f68e (2016-10-02): Better fix for breakage from c24a3150.

Closes #15.
closed 2016-10-02 by wdoekes