wdoekes/asterisk-chan-dongle #18: Cannot load chan_dongle.so

qpirsel, 2016-11-13:
Hi!

I'm currently trying to build a gsm gateway with freepbx and chan_dongle.  I've already spent a lot of time figuring out the error message **automake: error: no 'Makefile.am' found for any configure output** during **autoconf -a**.

Because I'm far away from whats being an expert I thought I can skip the next steps until I found this [post ](https://github.com/wdoekes/asterisk-chan-dongle/issues/15) and followed the procedure except with astversion 131202 which I'm running and got the module installed - although I've still no idea if I can ignore the autoconf error...

Anyway, after all I tried to show up my device in asterisk but it didn't know the dongle command. Then I realized that I should check if the module is already loaded but it wasn't so I tried it manually with the result shown below.

> root@orangepipc:/var/log/asterisk# asterisk -r
> Asterisk 13.12.2, Copyright (C) 1999 - 2014, Digium, Inc. and others.
> Created by Mark Spencer <markster@digium.com>
> Asterisk comes with ABSOLUTELY NO WARRANTY; type 'core show warranty' for details.
> This is free software, with components licensed under the GNU General Public
> License version 2 and other licenses; you are welcome to redistribute it under
> certain conditions. Type 'core show license' for details.
> =========================================================================
> Connected to Asterisk 13.12.2 currently running on orangepipc (pid = 1561)
> orangepipc*CLI> module load chan_dongle.so
> Unable to load module chan_dongle.so
> Command 'module load chan_dongle.so' failed.
> [2016-11-13 08:20:10] WARNING[2767]: loader.c:599 load_dynamic_module: Error loading module 'chan_dongle.so': /usr/lib/asterisk/modules/chan_dongle.so: undefined symbol: manager_event_sent_notify
> [2016-11-13 08:20:10] WARNING[2767]: loader.c:1087 load_resource: Module 'chan_dongle.so' could not be loaded.

Thanks in advance for any advise

wdoekes, 2016-11-13:
Works fine over here with Asterisk 13. Are you sure you checked out and built the *latest* version?

qpirsel, 2016-11-13:
I've performed a clean git clone https://github.com/wdoekes/asterisk-chan-dongle.git just now. When I configure with astversion make will abort as below:

```
make
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT app.o -MF .app.o.d -MP  -o app.o -c app.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT at_command.o -MF .at_command.o.d -MP  -o at_command.o -c at_command.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT at_parse.o -MF .at_parse.o.d -MP  -o at_parse.o -c at_parse.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT at_queue.o -MF .at_queue.o.d -MP  -o at_queue.o -c at_queue.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT at_read.o -MF .at_read.o.d -MP  -o at_read.o -c at_read.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT at_response.o -MF .at_response.o.d -MP  -o at_response.o -c at_response.c
In file included from at_response.c:25:0:
channel.h:32:16: warning: ‘struct ast_assigned_ids’ declared inside parameter list
   const struct ast_channel * requestor);
                ^
channel.h:32:16: warning: its scope is only this definition or declaration, which is probably not what you want
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT chan_dongle.o -MF .chan_dongle.o.d -MP  -o chan_dongle.o -c chan_dongle.c
In file included from chan_dongle.c:69:0:
channel.h:32:16: warning: ‘struct ast_assigned_ids’ declared inside parameter list
   const struct ast_channel * requestor);
                ^
chan_dongle.c: In function ‘public_state_init’:
chan_dongle.c:1660:59: error: ‘AST_FORMAT_CAP_FLAG_DEFAULT’ undeclared (first use in this function)
    if (!(channel_tech.capabilities = ast_format_cap_alloc(AST_FORMAT_CAP_FLAG_DEFAULT))) {
                                                           ^
chan_dongle.c:1660:59: note: each undeclared identifier is reported only once for each function it appears in
chan_dongle.c:1660:38: error: too many arguments to function ‘ast_format_cap_alloc’
    if (!(channel_tech.capabilities = ast_format_cap_alloc(AST_FORMAT_CAP_FLAG_DEFAULT))) {
                                      ^
In file included from /usr/include/asterisk/format_pref.h:28:0,
                 from /usr/include/asterisk/frame.h:34,
                 from /usr/include/asterisk/channel.h:141,
                 from ast_compat.h:10,
                 from chan_dongle.c:53:
/usr/include/asterisk/format_cap.h:57:24: note: declared here
 struct ast_format_cap *ast_format_cap_alloc(void);
                        ^
chan_dongle.c:1663:53: warning: passing argument 2 of ‘ast_format_cap_append’ from incompatible pointer type
    ast_format_cap_append(channel_tech.capabilities, ast_format_slin, 0);
                                                     ^
In file included from /usr/include/asterisk/format_pref.h:28:0,
                 from /usr/include/asterisk/frame.h:34,
                 from /usr/include/asterisk/channel.h:141,
                 from ast_compat.h:10,
                 from chan_dongle.c:53:
/usr/include/asterisk/format_cap.h:95:6: note: expected ‘const struct ast_format_cap *’ but argument is of type ‘struct ast_format *’
 void ast_format_cap_append(struct ast_format_cap *dst, const struct ast_format_cap *src);
      ^
chan_dongle.c:1663:4: error: too many arguments to function ‘ast_format_cap_append’
    ast_format_cap_append(channel_tech.capabilities, ast_format_slin, 0);
    ^
In file included from /usr/include/asterisk/format_pref.h:28:0,
                 from /usr/include/asterisk/frame.h:34,
                 from /usr/include/asterisk/channel.h:141,
                 from ast_compat.h:10,
                 from chan_dongle.c:53:
/usr/include/asterisk/format_cap.h:95:6: note: declared here
 void ast_format_cap_append(struct ast_format_cap *dst, const struct ast_format_cap *src);
      ^
In file included from chan_dongle.c:44:0:
chan_dongle.c: At top level:
chan_dongle.c:1775:1: error: unknown field ‘support_level’ specified in initializer
 AST_MODULE_INFO(ASTERISK_GPL_KEY, AST_MODFLAG_DEFAULT, MODULE_DESCRIPTION,
 ^
chan_dongle.c:1777:20: error: ‘AST_MODULE_SUPPORT_EXTENDED’ undeclared here (not in a function)
   .support_level = AST_MODULE_SUPPORT_EXTENDED,
                    ^
Makefile:66: recipe for target 'chan_dongle.o' failed
make: *** [chan_dongle.o] Error 1
root@orangepipc:/usr/src/asterisk-chan-dongle#
```

When I run **./configure --with-asterisk=/usr/src/asterisk-13.12.2/include** I get this:

```
[... cut ...]
checking host system type... armv7l-unknown-linux-gnueabihf
checking target system type... armv7l-unknown-linux-gnueabihf
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
configure: error: Please set --with-astversion=Mmmuu (major.minor.micro)

```

qpirsel, 2016-11-15:
K, I tried serveral variants including the one posted on the man page (always with clean clones). It doesn't make any difference. Ther error is always the same during make.

wdoekes, 2016-11-16:
You should not have to pass `./configure --with-asterisk=/usr/src/asterisk-13.12.2/include` if you're not cross compiling.

If you have installed your Asterisk correctly, its include files are in `/usr/include`. You will need to set `--with-astversion=Mmmuu` to the version that corresponds to the installed version.

Here you can see the canonical way to build chan_dongle, even with different versions:
https://github.com/wdoekes/asterisk-chan-dongle/blob/master/.travis.yml#L26

And here the latest build report for 13:
https://travis-ci.org/wdoekes/asterisk-chan-dongle/jobs/175508169

qpirsel, 2016-11-17:
Ok, the canonical way seems to work. At least I can load the module after placing the module and conf file to the correct places.

closed 2016-11-22 by qpirsel