wdoekes/asterisk-chan-dongle #20: Asterisk 14 for ubuntu 16.04

k008, 2016-11-19:
1. It is written in the instructions ./bootst**r**ap  and ./bootstap  should be
2. 
`make
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT app.o -MF .app.o.d -MP  -o app.o -c app.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT at_command.o -MF .at_command.o.d -MP  -o at_command.o -c at_command.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT at_parse.o -MF .at_parse.o.d -MP  -o at_parse.o -c at_parse.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT at_queue.o -MF .at_queue.o.d -MP  -o at_queue.o -c at_queue.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT at_read.o -MF .at_read.o.d -MP  -o at_read.o -c at_read.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT at_response.o -MF .at_response.o.d -MP  -o at_response.o -c at_response.c
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT chan_dongle.o -MF .chan_dongle.o.d -MP  -o chan_dongle.o -c chan_dongle.c
chan_dongle.c:37:23: error: expected declaration specifiers or ‘...’ before string constant
 ASTERISK_FILE_VERSION(__FILE__, "$Rev: " PACKAGE_REVISION " $")
                       ^
chan_dongle.c:37:33: error: expected declaration specifiers or ‘...’ before string constant
 ASTERISK_FILE_VERSION(__FILE__, "$Rev: " PACKAGE_REVISION " $")
                                 ^
Makefile:66: ошибка выполнения рецепта для цели «chan_dongle.o»
make: *** [chan_dongle.o] Ошибка 1
`

k008, 2016-11-19:
Fix
./configure --with-astversion=140000 \
        --with-asterisk=$HOME/src/asterisk-14/include
    make clean all

wdoekes, 2017-01-20:
Thanks for the report!

Note that --with-asterisk is not at all necessary if you installed Asterisk properly already.

I've updated the documentation.

closed 2017-01-20 by wdoekes