wdoekes/asterisk-chan-dongle #21: Strange sound problem with e1550 & e1552 usb modems

igorbgvdk, 2016-12-27:
Hi,

I have a strange problem with the chan_dongle setup. I have 10+ modems and on all have the same issue.
I have a quality hubs with external power, usb 2.0, I tried everything from Asterisk 1.8, 11, 13 with or without FreePBX with the latest version of asterisk-chan-dongle. I tried RasPBX which had the same issues.

Issue is related only to the incoming calls, once the .Answer() function is called from the dial plan, on 75% of calls .Answer() will produce this sound on the caller end (also sound is captured in call recording on asterisk): https://clyp.it/53m02bxk .

igorbgvdk, 2016-12-30:
The same problem with e153, e160, e173?
I tried also e3251, doesn't create the same issue (but have some issues related to volume, etc...).

To be honest, this is OK to play around for day or two, nothing more until this is resolved....

wdoekes, 2017-01-20:
Thanks for the report. Note that I do not run this software, nor do I own a dongle. So I can be of limited help in fixing this.

If come up with a fix, I'll be happy to review and merge.

gamba47, 2017-01-27:
Sounds like an external sound not asterisk o chan_dongle generated by itself.

Best regards.

wdoekes, 2019-11-20:
Stale issue. I cannot help with this without more info.

closed 2019-11-20 by wdoekes