wdoekes/asterisk-chan-dongle #23: Support ZTE modems

k008, 2017-01-21:
Hello!
Is it possible to add to the list of supported modems - **ZTE**?

wdoekes, 2017-01-21:
I have no idea. I have no dongle, let alone a ZTE dongle.

If you provide a PR, I'll happily review it for sanity and merge it.

k008, 2017-01-21:
PR? - who is this / that?
I have ZTE MF180, if I had the opportunity to get his team and send them to you ...

gamba47, 2017-01-27:
You need a Modem with voice enabled AT commands and unlocked to make it possible.
those modems don't have this functionality enabled.

The begginig of this theme is here https://en.wikipedia.org/wiki/Voice_modem_command_set
Every modem manufacturer don't show us your at commands and this is the main problem.

Best regards!
gamba47