wdoekes/asterisk-chan-dongle #26: opus codec problem

lnx13, 2017-02-06:
There are no outgoing sound with opus 48kHz codec on asterisk 14.
```
[Feb  3 19:19:17] DEBUG[22697][C-00000003]: channel.c:5657 set_format: Channel Dongle/dongle0-0100000001 setting read format path: slin -> slin48
[Feb  3 19:19:17] DEBUG[22697][C-00000003]: channel.c:5657 set_format: Channel SIP/9912-00000002 setting write format path: slin48 -> opus
[Feb  3 19:19:17] DEBUG[22697][C-00000003]: channel.c:5657 set_format: Channel SIP/9912-00000002 setting read format path: opus -> slin48
[Feb  3 19:19:17] DEBUG[22697][C-00000003]: channel.c:5657 set_format: Channel Dongle/dongle0-0100000001 setting write format path: slin48 -> slin
```

wdoekes, 2017-02-15:
I have no idea. I have no dongle, let alone a E1550 dongle.

If you provide a PR, I'll happily review it for sanity and merge it.

alaa2003, 2017-08-10:
hello my friend
when VOIP sound equal skype quality that means great achieve 
i am interested with great voice with OPUS codec to work with chan_dongle
currently i am using RPI, In order to get opus working on RPI, it should be compiled due to arm cpu
Andrius Kairiukstis noted me armhf architecture may be not good and better using Intel Atom
raspbx had limitation and strange issues like suddenly dongles disconnect, might chan_dongle bugs
also i was asking my self if OPUS has dynamic switch between its bands depending on internet speed ( like skype / viber )
asterisk gives OPUS free from asterisk 14 , i think lately they updated on 13 too
http://downloads.digium.com/pub/telephony/codec_opus/
i am happy your chan_dongle works with asterisk 14 :)
here important links of OPUS:
http://info.digium.com/index.php/email/emailWebview?mkt_tok=eyJpIjoiTkdKak0yRTRZelZrTXpJeiIsInQiOiJ6Nzl1bUxGSE1ZZGI4S2tpK2ZvTFI2blp6Qk81WXlCSEtyWlp1U0lFajZoQURmUVlETkl2cFNoUU5jSXZoRHpXNk1DeHl5NEFvTU5TRE1EV1YyMUVSYlUzRk5EN09QcjZKeWVKZEU0XC9vRzZJck9wMmhsUEhPWWdmNmoyZ1NMeWoifQ%3D%3D

http://blogs.asterisk.org/2017/04/12/asterisk-opus-packet-loss-fec/?utm_source=digium_em&utm_medium=email&utm_campaign=AS_NL_062017&mkt_tok=eyJpIjoiTkdKak0yRTRZelZrTXpJeiIsInQiOiJ6Nzl1bUxGSE1ZZGI4S2tpK2ZvTFI2blp6Qk81WXlCSEtyWlp1U0lFajZoQURmUVlETkl2cFNoUU5jSXZoRHpXNk1DeHl5NEFvTU5TRE1EV1YyMUVSYlUzRk5EN09QcjZKeWVKZEU0XC9vRzZJck9wMmhsUEhPWWdmNmoyZ1NMeWoifQ%3D%3D

https://wiki.asterisk.org/wiki/display/AST/Codec+Opus

http://blogs.digium.com/2016/09/30/opus-in-asterisk/?utm_source=digium_em&utm_medium=email&utm_campaign=AS_NL?utm_source=digium_em&utm_medium=email&utm_campaign=AS_NL_062017&mkt_tok=eyJpIjoiTkdKak0yRTRZelZrTXpJeiIsInQiOiJ6Nzl1bUxGSE1ZZGI4S2tpK2ZvTFI2blp6Qk81WXlCSEtyWlp1U0lFajZoQURmUVlETkl2cFNoUU5jSXZoRHpXNk1DeHl5NEFvTU5TRE1EV1YyMUVSYlUzRk5EN09QcjZKeWVKZEU0XC9vRzZJck9wMmhsUEhPWWdmNmoyZ1NMeWoifQ%3D%3D

http://www.circleid.com/posts/200130727_why_the_opus_codec_matters_even_if_you_dont_care_about_audio/

happy reading :)

wdoekes, 2019-11-20:
Stale issue. I cannot help with this without PRs.

closed 2019-11-20 by wdoekes