wdoekes/asterisk-chan-dongle #27: installation question

Nufay, 2017-02-15:
Hi, i wanted to ask you about one issue, which, i think, you know how to deal with.

I ran into a problem on compiling an abandoned fork of chan_dongle for CDMA dongles.
I'm getting this error when trying to do checkinstall:

```
========================= Installation results ===========================
gcc -g -O2 -O6 -I.  -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT app.o -MF .app.o.d -MP  -o app.o -c app.c
In file included from app.c:19:0:
/usr/include/asterisk.h:300:2: error: #error "Externally compiled modules must declare AST_MODULE_SELF_SYM."
 #error "Externally compiled modules must declare AST_MODULE_SELF_SYM."
  ^
In file included from app.c:23:0:
/usr/include/asterisk/version.h:1:2: error: #error "Do not include 'asterisk/version.h'; use 'asterisk/ast_version.h' instead."
 #error "Do not include 'asterisk/version.h'; use 'asterisk/ast_version.h' instead."
  ^
app.c: In function ‘app_register’:
app.c:164:43: warning: passing argument 2 of ‘ast_register_application2’ from incompatible pointer type
   ast_register_application2 (dca[i].name, (app_func_t)(dca[i].func), dca[i].synopsis, dca[i].desc, self_module());
                                           ^
In file included from app.c:22:0:
/usr/include/asterisk/module.h:608:5: note: expected ‘int (*)(struct ast_channel *, const char *)’ but argument is of type ‘int (*)(struct ast_channel *, void *)’
 int ast_register_application2(const char *app, int (*execute)(struct ast_channel *, const char *),
     ^
Makefile:67: recipe for target 'app.o' failed
make: *** [app.o] Error 1

****  Installation failed. Aborting package creation.

Cleaning up...OK

Bye.
```

Can you, please, point me in a direction of actions to do?

Btw standard chan_dongle installation process ends same, but i found your fork and now my Huawei dongle works excellent, thank you for this!
Best regards,
Nufay.

wdoekes, 2017-02-15:
```
asterisk-chan-dongle$ git log -S AST_MODULE_SELF_SYM
commit 9529a68056331e46fceca8a3480c5883d931e1f1
Author: Walter Doekes <walter+github@wjd.nu>
Date:   Sun Sep 4 16:19:28 2016 +0200

    Fix compile issues with Asterisk 14.
    
    Declare AST_MODULE_SELF_SYM=__internal_$(module_name)_self.
```

Nufay, 2017-02-16:
After applying your changes to my files one of errors disappeared, but other left unsolved:

```
========================= Installation results ===========================
gcc -g -O2 -O6 -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT app.o -MF .app.o.d -MP  -o app.o -c app.c
In file included from app.c:23:0:
/usr/include/asterisk/version.h:1:2: error: #error "Do not include 'asterisk/version.h'; use 'asterisk/ast_version.h' instead."
 #error "Do not include 'asterisk/version.h'; use 'asterisk/ast_version.h' instead."
  ^
app.c: In function ‘app_register’:
app.c:164:43: warning: passing argument 2 of ‘ast_register_application2’ from incompatible pointer type
   ast_register_application2 (dca[i].name, (app_func_t)(dca[i].func), dca[i].synopsis, dca[i].desc, self_module());
                                           ^
In file included from app.c:22:0:
/usr/include/asterisk/module.h:608:5: note: expected ‘int (*)(struct ast_channel *, const char *)’ but argument is of type ‘int (*)(struct ast_channel *, void *)’
 int ast_register_application2(const char *app, int (*execute)(struct ast_channel *, const char *),
     ^
Makefile:68: recipe for target 'app.o' failed
make: *** [app.o] Error 1

****  Installation failed. Aborting package creation.

Cleaning up...OK

Bye.

```

Any hints?
Regards, 
Nufay.

wdoekes, 2017-02-16:
```
$ git log -S asterisk/version.h
commit 9108a966aa146487815f639f653cd0401e2849c5
Author: Walter Doekes <walter+github@wjd.nu>
Date:   Sat Sep 5 17:21:25 2015 +0200

    Add Asterisk 11 support.
    
    Author: me, in August 2013.

```

```
$ git show 9108a966aa146487815f639f653cd0401e2849c5 | grep -F /version.h
-#include <asterisk/version.h>	/* ASTERISK_VERSION_NUM */
-#include <asterisk/version.h>
-#include <asterisk/version.h>			/* ASTERISK_VERSION_NUM */
-#include <asterisk/version.h>			/* ASTERISK_VERSION_NUM */
```

Nufay, 2017-02-16:
I'm a little bit confused now, do you mean i need to change my code according to your's
_commit 9108a966aa146487815f639f653cd0401e2849c5_ ?

I superficially read _**channel.c**_ file and there is nothing about asterisk versions higher than 1.8, and it got twice as much lines comparing with yours in this commit.
So, is there any sense editing those files or it'll be better to install older version of asterisk?

wdoekes, 2017-02-16:
I'm saying that:
- you have a compilation problem with version.h
- I fixed a problem with version.h in the commit which you can find using `git log -S version.h`

Whether or not you want or need to apply all those fixes or just some is up to you.

Alternatively, you can go a different route, and check the differences between standard chan_dongle and CDMA-chan_dongle and apply *those* differences onto this chan_dongle.

Nufay, 2017-02-16:
Any hints for this kind of error?

```
========================= Installation results ===========================
./config.status
config.status: creating Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
gcc -g -O2 -O6 -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT chan_dongle.o -MF .chan_dongle.o.d -MP  -o chan_dongle.o -c chan_dongle.c
chan_dongle.c:44:1: error: expected declaration specifiers or ‘...’ before string constant
 ASTERISK_FILE_VERSION(__FILE__, "$Rev: " PACKAGE_REVISION " $")
 ^
chan_dongle.c:44:33: error: expected declaration specifiers or ‘...’ before string constant
 ASTERISK_FILE_VERSION(__FILE__, "$Rev: " PACKAGE_REVISION " $")
                                 ^
In file included from chan_dongle.c:66:0:
dmlib.h:70:1: warning: useless storage class specifier in empty declaration
 } __attribute__ ((packed));
 ^
Makefile:68: recipe for target 'chan_dongle.o' failed
make: *** [chan_dongle.o] Error 1

****  Installation failed. Aborting package creation.

Cleaning up...OK

Bye.

```

wdoekes, 2017-02-16:
```
$ git show 53e0be53
commit 53e0be53a097087cfc34ba4974027303fe80ac32
Author: Walter Doekes <walter+github@wjd.nu>
Date:   Tue May 10 22:02:24 2016 +0200

    Fix compile issues with master (Asterisk 14).

diff --git a/chan_dongle.c b/chan_dongle.c
index ff87e32..b18d143 100644
--- a/chan_dongle.c
+++ b/chan_dongle.c
@@ -36,7 +36,10 @@
 #endif /* HAVE_CONFIG_H */
 
 #include <asterisk.h>
+
+#if ASTERISK_VERSION_NUM < 140000 /* 14- */
 ASTERISK_FILE_VERSION(__FILE__, "$Rev: " PACKAGE_REVISION " $")
+#endif /* 14- */
 
 #include <asterisk/stringfields.h>     /* AST_DECLARE_STRING_FIELDS for asterisk/manager.h */
 #include <asterisk/manager.h>
diff --git a/cli.c b/cli.c
index b13e222..71468da 100644
--- a/cli.c
+++ b/cli.c
@@ -933,6 +933,10 @@ static struct ast_cli_entry cli[] = {
        AST_CLI_DEFINE (cli_discovery,          "Discovery devices and create config"),
 };
 
+#if ASTERISK_VERSION_NUM >= 140000 /* 14+ */
+# define AST_MODULE_SELF self_module()
+#endif /* 14+ */
+
 #/* */
 EXPORT_DEF void cli_register()
 {
```

Nufay, 2017-02-17:
I'm sorry if i'm bothering you, but there's another error: 

```
========================= Installation results ===========================
./config.status
config.status: creating Makefile
config.status: creating config.h
config.status: config.h is unchanged
config.status: executing depfiles commands
gcc -g -O2 -O6 -I. -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT chan_dongle.o -MF .chan_dongle.o.d -MP  -o chan_dongle.o -c chan_dongle.c
In file included from chan_dongle.c:68:0:
dmlib.h:70:1: warning: useless storage class specifier in empty declaration
 } __attribute__ ((packed));
 ^
chan_dongle.c:78:30: error: storage size of ‘chan_dongle_format’ isn’t known
 EXPORT_DEF struct ast_format chan_dongle_format;
                              ^
Makefile:68: recipe for target 'chan_dongle.o' failed
make: *** [chan_dongle.o] Error 1

****  Installation failed. Aborting package creation.

Cleaning up...OK

Bye.

```

wdoekes, 2017-02-17:
```
$ git blame chan_dongle.h | grep chan_dongle_format -C2 | head -n5
7eb9b856 (Walter Doekes  2015-09-06 11:25:48 +0200  42) #if ASTERISK_VERSION_NUM >= 100000 && ASTERISK_VERSION_NUM < 130000 /* 10-13 */
aebd0734 (Walter Doekes  2015-09-05 17:03:52 +0200  43) /* Only linear is allowed */
aebd0734 (Walter Doekes  2015-09-05 17:03:52 +0200  44) EXPORT_DECL struct ast_format chan_dongle_format;
aebd0734 (Walter Doekes  2015-09-05 17:03:52 +0200  45) EXPORT_DECL struct ast_format_cap * chan_dongle_format_cap;
7eb9b856 (Walter Doekes  2015-09-06 11:25:48 +0200  46) #endif /* ^10-13 */
$ git blame chan_dongle.c | grep chan_dongle_format -C1 | head -n4
7eb9b856 (Walter Doekes  2015-09-06 11:25:48 +0200   75) #if ASTERISK_VERSION_NUM >= 100000 && ASTERISK_VERSION_NUM < 130000 /* 10-13 */
aebd0734 (Walter Doekes  2015-09-05 17:03:52 +0200   76) EXPORT_DEF struct ast_format chan_dongle_format;
aebd0734 (Walter Doekes  2015-09-05 17:03:52 +0200   77) EXPORT_DEF struct ast_format_cap * chan_dongle_format_cap;
7eb9b856 (Walter Doekes  2015-09-06 11:25:48 +0200   78) #endif /* ^10-13 */
```