wdoekes/asterisk-chan-dongle #28: Error parsing incoming messages: unsupported DCS value

marcogaio, 2017-03-04:
Sometimes i got:
`Mar  4 12:04:53 rita asterisk[3481]: WARNING[4677]: at_response.c:1228 in at_response_cmgr: [dongle0] Error parsing incoming message '+CMGR: 0,,137#015#01207919333851805320409D034186C360300F0713040214084408849A7F1099A36A72018ED059BB14031990E46D38186EF39FD0D1AA3D3E176981E06155D20182B177381926CD0585E26A7E96F10015474816839960CE70241CB7250DA6D7E83E67550D95E76D3EB61761AF486EBD36F771A14A6D3D3F632A80C12BFDDF539485E9EA7C9F534688C4E87DB61100D968BD95C' at possition 49: Unsupported DCS value`

If i try to parse the PDU with some online converters, i note that correctly parse.

dfgg12, 2017-03-05:
Try editing at_response.c and add you modem model then observe if parsing errors happen again.
https://github.com/wdoekes/asterisk-chan-dongle/blob/master/at_response.c#L1615

marcogaio, 2017-03-09:
OK, i've recompiled adding model 'K3765'.

But... how can i test it? If i try to resend (with my android phone) the exactly same message (as textual representation), they work but also worked with the unpatched chan-dongle version.

I need to resend the same PDU (or a PDU of the same type). How can i do?

Thanks.

wdoekes, 2017-03-09:
I don't think changing the model works indeed.

It trips over this:

```
TP-OA:	09D034186C3603
TP-PID:	00
TP-DCS:	F0
```
Up to the PID, it's fine, but at DCS `F0`, it should decode the message according to:
https://en.wikipedia.org/wiki/Data_Coding_Scheme
`F0	240	GSM 7 bit	Class 0 (Flash message)`

The code expects this to be true:
```PDU_DCS_76(dcs) == PDU_DCS_76_00```
but dsc (0xF0) & 0xC0 is not 0.

The code assumes this:

11000000 = C0 = "DSC76"
00100000 = 20 = COMPRESSION
00010000 = 10 = "BITS10" (unused)
00001100 = 0C = alphabet (0=7bit, 1=8bit, 2=usc2, 3=undefined)
00000011 = 03 = class (0=Flash message, 1=ME-specific, 2=SIM/USIM-specific, 3=TE-specific)

However, it looks like the values all depend on each other, and with DSC bits 7..4 set to `Fx`, they should be treated more or less like `0x`.

See also: http://www.etsi.org/deliver/etsi_gts/03/0338/05.00.00_60/gsmts_0338v050000p.pdf
Page 8: if bits 7..4 are F0, then:
```
"Data coding/message class"
Bit 3 is reserved, set to 0.
Bit 2 Message coding:
  0 Default alphabet
  1 8-bit data
Bit 1 Bit 0 Message Class:
  0 0 Class 0
  0 1 Class 1 default meaning: ME-specific.
  1 0 Class 2 SIM-specific message.
  1 1 Class 3 default meaning: TE specific (see GSM TS 07.05)
```

I don't know what the "data coding / message storage" class means, but it looks like that fact could be discarded and the message can be treated as a regular message.

wdoekes, 2017-03-09:
See also typos in BITS10:
```
--- a/pdu.c
+++ b/pdu.c
@@ -278,9 +278,9 @@
 
 /*   bit 4 */
 #define PDU_DCS_BITS10_CTRL_SHIFT              4
-#define PDU_DCS_BITS10_RETAIN                  (0x00 << PDU_DCS_BIT10_CTRL_SHIFT)
-#define PDU_DCS_BITS10_INUSE                   (0x01 << PDU_DCS_BIT10_CTRL_SHIFT)
-#define PDU_DCS_BITS10_CTRL_MASK               (0x01 << PDU_DCS_BIT10_CTRL_SHIFT)
+#define PDU_DCS_BITS10_RETAIN                  (0x00 << PDU_DCS_BITS10_CTRL_SHIFT)
+#define PDU_DCS_BITS10_INUSE                   (0x01 << PDU_DCS_BITS10_CTRL_SHIFT)
+#define PDU_DCS_BITS10_CTRL_MASK               (0x01 << PDU_DCS_BITS10_CTRL_SHIFT)
 #define PDU_DCS_BITS10_CTRL(dcs)               ((dcs) & PDU_DCS_BITS10_CTRL_MASK)
 
 /*   bit 5 */
```

marcogaio, 2017-03-10:
Patch applied. But:

1) (with or without patch applied) seems that sending USSD does not work any more:
`eraldo*CLI> dongle ussd dongle0 *123#
[dongle0] USSD queued for send with id 0x7f515001b9d0
[Mar 10 14:13:44] ERROR[29667]: at_response.c:517 at_response_error: [dongle0] Error sending USSD 0x7f515001b9d0
`
I use rarely USSD, but i remember that worked.

2) How we can test if worked or not, apart waiting the operator to send an USSD response with 7bit character set?

Thanks.

marcogaio, 2017-03-23:
OK, decoding errors still happens:

`Mar 23 18:10:55 rita asterisk[6596]: WARNING[11228]: at_response.c:1228 in at_response_cmgr: [dongle0] Error parsing incoming message '+CMGR: 0,,137#015#01207919333851805320409D034186C360300F0713032810105408849A7F1099A36A720D9EC059BB140319C2E06D38186EF39FD0D1AA3D3E176981E06155D20184B467381926CD0585E26A7E96F1001547481683816ACE60241CB7250DA6D7E83E67550D95E76D3EB61761AF486EBD36F771A14A6D3D3F632A80C12BFDDF539485E9EA7C9F534688C4E87DB61100D968BD95C' at possition 49: Unsupported DCS value`

`Mar 23 19:18:36 rita asterisk[6596]: WARNING[11228]: at_response.c:1228 in at_response_cmgr: [dongle0] Error parsing incoming message '+CMGR: 0,,137#015#01207919333851805320409D034186C360300F0713032918123408849A7F1099A36A720D9EC059BB140B19C2E86D38186EF39FD0D1AA3D3E176981E06155DA018CB367381926CD0585E26A7E96F10015474816836164DE60241CB7250DA6D7E83E67550D95E76D3EB61761AF486EBD36F771A14A6D3D3F632A80C12BFDDF539485E9EA7C9F534688C4E87DB61100D968BD95C' at possition 49: Unsupported DCS value`

I'm not a real programmer, especially in C, but if i can be useful in some way... say me...

marcogaio, 2017-04-07:
Some news?!

AFAI remember well, USSD (sending and receiving) worked in ''plain'' chan-dongle with asterisk 1.8, but as just stated i'm not a skilled C programmer and i'm not able to go back into commits to found the culprit... but i can easily apply patch and recompile, to test against.

Say me. Thanks.

wdoekes, 2017-04-16:
(This is not fixed, github thought it was because of "fix" in the commit msg.)

wdoekes, 2017-04-16:
This should fix things. Please try it.

marcogaio, 2017-04-17:
Still sending USSD does not work:
`eraldo*CLI> dongle ussd dongle0 *123#
[dongle0] USSD queued for send with id 0x7f659401b790
[Apr 18 00:07:43] ERROR[3452]: at_response.c:517 at_response_error: [dongle0] Error sending USSD 0x7f659401b790`

And still this is the only way i know to get back a flash message, so test error on subject. ;-(

marcogaio, 2017-04-17:
Ahem, forgot to write.

Say me if it is better to open a different bugreport, or better to reopen that...

wdoekes, 2017-04-18:
Oh, wait. Sending USSD is beyond the scope here.

Please file another bug report for that.

closed 2017-04-18 by wdoekes

marcogaio, 2017-04-18:
OK. Added issue #29 .

PS: anyway, your code compile flawlessy and seems to work well, simply i've to wait the operator to send me flash messages to test it, and i don't know how without sending it an USSD command.

commit fba6ed16 (2017-04-16): pdu/tests: Clean up pdu_parse if-tree and restore test capability.

    make check

to build and run tests.

Cleanups were required to even begin attempting to fix #28.
commit 9c100815 (2017-04-16): pdu/parsing: Fix unsupported DCS value 0xF0.

Closes #28.