wdoekes/asterisk-chan-dongle #30: segfault at commit 14a966d

laigor, 2017-04-18:
When build and loading module chan_dongle
asterisk[307]: segfault at 0 ip 00007f9c9b1a2ddb sp 00007f9c9b4409e0 error 4 in chan_dongle.so[7f9c9b193000+30000
debian stretch amd64
Asterisk 13.14.1~dfsg-1

When build 48df4f606a0b7ea0aa67daec5090be45f004d3e9 no segfault

wdoekes, 2017-04-18:
Interesting.

Can you fetch a code location from addr2line?
```
addr2line -Cfe PATH/TO/chan_dongle.so 0xFDDB
# (ip (00007f9c9b1a2ddb) - offset (7f9c9b193000))
```
Can you pin it down to a specific commit? (See: `git bisect`)
Can you get a backtrace from gdb?

laigor, 2017-04-18:
7c8e6b8210aeeb4d2c622e27650b390d0128dd25 latest good commit

wdoekes, 2017-04-18:
I think that should fix things. Sorry for the breakage. Thanks for the report :)

commit 45d8fbc6 (2017-04-18): cleanup/fix: Undo pvt=NULL "fix" which was bad.

Fixes #30 I think.
closed 2017-04-18 by wdoekes