wdoekes/asterisk-chan-dongle #32: chan_dongle.so: undefined symbol: ast_verb_sys_level

k008, 2017-05-17:
Hello!
Did not understand then, the module stopped working.
In the Asterisk console:
```
module load chan_dongle.so
Unable to load module chan_dongle.so
Command 'module load chan_dongle.so' failed.
[May 17 15:33:52] WARNING[17037]: loader.c:396 load_dynamic_module: Error loading module 'chan_dongle.so': /usr/lib/asterisk/modules/chan_dongle.so: undefined symbol: ast_verb_sys_level
[May 17 15:33:52] WARNING[17037]: loader.c:848 load_resource: Module 'chan_dongle.so' could not be loaded.
```

wdoekes, 2017-05-17:
Are you trying to run an old compiled chan_dongle with a newer Asterisk 11+ ?

k008, 2017-05-17:
I'm not going to say exactly, and yes, version 13 as well.
Perhaps the update Ubuntu broke something ...

Today I downloaded the latest version from you, compiled it and got the same.
Asterisk did not reinstall

wdoekes, 2017-05-18:
I'm having a hard time understanding what you're saying exactly.

However, I do understand that it still doesn't work.

Please check everything and report back:
- your current asterisk version
- the build method (`--with-astversion=...`)
- where your modules are according to asterisk.conf
- where your chan_dongle.so module is

k008, 2017-05-18:
Exactly the version of Asterisk could not find out (is it when compiling the module?): 131200 - previously worked and 13.7 - freshly compiled

/etc/asterisk/asterisk.conf
/usr/lib/asterisk/modules/chan_dongle.so

wdoekes, 2017-05-18:
> when compiling the module?

Eh.. what?

The module version is supposed to match the asterisk version. So if you have Asterisk 13.7.0 installed, you should `./configure --with-astversion=13.7.0`

If you cannot find out what your currently running version of Asterisk is, I will not be able to help you.

k008, 2017-05-18:
asterisk_13.12.2-1_i386

wdoekes, 2017-05-18:
So, you've compiled the chan_dongle locally? With `--with-astversion=13.12.2` and you've double checked that the module is in the right `modules/` directory?

k008, 2017-05-18:
`/usr/src/asterisk-chan-dongle-master# ./configure --with-astversion=13.12.2 --with-asterisk=/usr/src/asterisk-13.12.2/include
checking build system type... i686-pc-linux-gnu
checking host system type... i686-pc-linux-gnu
checking target system type... i686-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
./configure: line 2905: 13.12.2: syntax error: invalid arithmetic operator (error token is ".12.2")`

k008, 2017-05-18:
`./configure --with-astversion=131200 --with-asterisk=/usr/src/asterisk-13.12.2/include/`
Success compile, but not work

wdoekes, 2017-05-18:
(1) You shouldn't need to include the `--with-asterisk` path. Instead, you need to install the asterisk include files into the `/usr/include` by `make install` when installing asterisk. Or by installing `asterisk-dev(el)` if you're using a packaged build.

(2) As for `invalid arithmetic operator (error token is ".12.2")`. You're not running the latest version of chan_dongle. You said:
> Today I downloaded the latest version from you, compiled it and got the same.
That's not the latest version.

```
commit 8bbba90ec2d5de7214705395b923e12efdb73a86
Date:   Fri Jan 20 11:59:53 2017 +0100

    build: Improve --with-astversion to take X.Y.Z version.
    
    This should clarify a few things.
```

k008, 2017-05-18:
1. I compile Asterisk:
` 
./configure
make menuselect
contrib/scripts/get_mp3_source.sh
checkinstall
`
In / usr / include no asterisk.h

2. I do not understand why I do not have the latest version and where can I get it?
And how do I specify the version of Asterisk 13.12.2?

wdoekes, 2017-05-18:
I don't know what your checkinstall does, but the regular install works:
```
asterisk-rw-13.git$ make install
...
/usr/bin/install -c -m 644 include/asterisk.h "/usr/include"
/usr/bin/install -c -m 644 include/asterisk/*.h "/usr/include/asterisk"
/usr/bin/install -c -m 644 include/asterisk/doxygen/*.h "/usr/include/asterisk/doxygen"
...
```

Compiling chan_dongle:
```
asterisk-chan-dongle$ ./configure --with-astversion=13.15
...

asterisk-chan-dongle$ make

asterisk-chan-dongle$ sudo make install
strip chan_dongle.so
/usr/bin/install -c -m 755 chan_dongle.so /usr/lib/asterisk/modules
```

Loading it:
```
$ sudo touch /etc/asterisk/dongle.conf

$ sudo asterisk -c
Asterisk GIT-13-13.15.0-rc1-153-g722ec0671e, Copyright (C) 1999 - 2014, Digium, Inc. and others.

*CLI> module load chan_dongle.so
Loaded chan_dongle.so

*CLI> dongle show version 

chan_dongle: Huawei 3G Dongle Channel Driver, Version 1.1, Revision 43gh=47cg
Project Home: https://github.com/bg111/asterisk-chan-dongle/
Bug Reporting: https://github.com/bg111/asterisk-chan-dongle/issues

k008, 2017-05-18:
Checkinstall collects into a package.
To specify the path to asterisk is not an error?
I get the same error
```
 Module load chan_dongle.so
Unable to load module chan_dongle.so
Command 'module load chan_dongle.so' failed.
[May 18 13:29:03] WARNING [19648]: loader.c: 396 load_dynamic_module: Error loading module 'chan_dongle.so': /usr/lib/asterisk/modules/chan_dongle.so: undefined symbol: ast_verb_sys_level
[May 18 13:29:03] WARNING [19648]: loader.c: 848 load_resource: Module 'chan_dongle.so' could not be loaded.
```
, With:
`./configure --with-astversion = 13.12.2 --with-asterisk = / usr / src / asterisk-13.12.2 / include`

wdoekes, 2017-05-18:
Well, from what I've seen, the problem is entirely with your configuration.

Until you stop doing non-standard stuff, I don't think anyone can help you. You'll have to play around to better understand where your problem is. Perhaps reinstall everything from scratch. Perhaps use a packaged version. Perhaps take a better look at any warning/notice during compilation of both asterisk and chan-dongle. Perhaps downgrade and check if that works, and then work your way up to see when the problems start.

k008, 2017-05-18:
Six months ago, everything worked!
What could have changed?
What non-standard did you notice? Checkinstall?

wdoekes, 2017-05-19:
> Perhaps downgrade and check if that works, and then work your way up to see when the problems start.

Until you come up with more information, there is nothing you can reasonably expect me to do.

wdoekes, 2017-07-05:
No reporter feedback. Closing out.

closed 2017-07-05 by wdoekes

k008, 2017-07-10:
Thank you, you were right. I reinstalled Asterisk and your module. Everything worked.
Note. I installed it, as described above, using my deb files.
Now for ubuntu there are updates for Asterisk, there will be a time - I will update and check the performance, maybe this all broke me.