wdoekes/asterisk-chan-dongle #37: Discovery by imei problem.

laigor, 2017-07-17:
in dongle.conf:
[dongle0]
imei=867455001874393

[dongle3]
imei=867455001874393

All modems huawei e173
asterisk console:
[Jul 17 15:58:26] ERROR[18905]: at_response.c:118 at_response_ok: [dongle0] Received unexpected 'OK'
[Jul 17 15:58:28] ERROR[18943]: chan_dongle.c:440 do_monitor_phone: [dongle3] timedout while waiting 'OK' in response to 'AT'
[Jul 17 16:08:20] ERROR[18942]: at_response.c:118 at_response_ok: [dongle3] Received unexpected 'OK'
[Jul 17 16:08:22] ERROR[18904]: pdiscovery.c:570 pdiscovery_do_cmd: [dongle0 discovery] failed to get valid response from /dev/ttyUSB8 in 500 msec
It seems the messages destined for the dongle3 are sent to the wrong com port...

wdoekes, 2017-07-24:
Too little info. I can't deduce any cause from these error messages alone.

CpServiceSpb, 2017-08-20:
If I disable or remove 
audio=/dev/......
data=/dev/......
and add or enable 
imei = ...............
I will not my dongle connected to dongle chan.
That is dongle show devices is empty.

gamba47, 2017-08-28:
When you start asterisk without any dongle configured, you can search for a device with this command:

```
Asterisk *CLI> dongle discovery 
; discovered device
[dc_0177_0724](defaults)
;audio=/dev/ttyUSB1
;data=/dev/ttyUSB2
imei=35921103712xxxx
imsi=72234124377xxxx

```

As you can see, there is "imei" but "imsi" is important too. Those lines show a common configuration for dongle.conf. Showing the actual ttyUSB# used by audio and data.

You need to put both lines in your dongle.conf

Best regards.

Emiliano - gamba47

oooo1, 2017-08-30:
How is to specify /dev/............. ?
I reassigned /dev/ttyUSB9 to /dev/devModem.
Even not /dev/ttyUSB1.
But dongle discovery requests /dev/ttyUSB1 only.

gamba47, 2017-08-31:
Post the output of this commands (as root):

* lsusb
* asterisk -rx "dongle show devices"
* asterisk -rx "dongle discovery"
* cat /etc/asterisk/dongle.conf

best regards.

gamba47, 2017-09-04:
Ok, dongle discovery only show new and unconfigured modems at your server.

Please do this:
1. Edit dongle.conf and comment  audio and data lines (the last two lines of this)
2. restart asterisk
3. check "asterisk -rx "dongle show devices". You need to see this empty
4. Then run "asterisk -rx "dongle discovery" and you will see your IMEI and IMSI.

Best regards.

Emiliano -

oooo1, 2017-09-06:
Nothing changed.

asterisk -rx "dongle show devices"
`ID           Group State      RSSI Mode Submode Provider Name  Model      Firmware          IMEI             IMSI             Number
`

 asterisk -rx "dongle discovery"
```
; discovered device
[dc__](defaults)
;audio=/dev/ttyUSB2
;data=/dev/ttyUSB1
imei=
imsi=
```