wdoekes/asterisk-chan-dongle #38: Error decode CUSD

c-e-p-x-u-o, 2017-08-11:
cellgate*CLI> dongle ussd tele2 *105#
[tele2] USSD queued for send with id 0xb4700660
    -- [tele2] Successfully sent USSD 0xb4700660
[Aug 11 03:26:55] NOTICE[17367]: at_response.c:266 at_response_ok: [tele2] Successfully sent USSD 0xb4700660
[Aug 11 03:26:56] ERROR[17367]: at_response.c:1384 at_response_cusd: [tele2] Error decode CUSD: CF2135487D2E4131DCCD25A381E02E

But http://smstools3.kekekasvi.com/topic.php?id=288 can successfully convert this CUSD

c-e-p-x-u-o, 2017-08-11:
other ussd works:

cellgate*CLI> dongle ussd tele2 *201#
[tele2] USSD queued for send with id 0x8080bd78
    -- [tele2] Successfully sent USSD 0x8080bd78
[Aug 11 04:04:22] NOTICE[18775]: at_response.c:266 at_response_ok: [tele2] Successfully sent USSD 0x8080bd78
 [tele2] Got USSD type 0 'USSD Notify': 'Ваш федеральный номер +7 (999) 666-13-13'

CpServiceSpb, 2017-08-20:
I confirm this issue.
More over I didn' t get *201# working as well.
And in many USSD cases I got:
CE325C1EB6A7D9EE7C1AE47EB7CB72507DCEAE9FD3
That is meant "Nepravilniy nomer usugi" - it is English for "Wrong service number" .
I made some discovering of the issue and found out, that it happens because of Huawei has to send USSD in PDU format and receives answer in text format. But before sending, USSD request is not encoded to PDU.
Moreover I made some changes for some previous version of wdoekes chan dongle version and got working USSD correctly.
But, unfortunately I lost those version.

CpServiceSpb, 2017-08-20:
I remembered some part of how I solved it.
Though my Huawei E367 is set up for PDU mode (AT+CMGF=0) outgoing USSD were not encoded to PDU format for sending.
I don' t know is it right or not, but I didn' t change dongle mode but I added "E367" to seven_bit_modems array to at_response.c.
And, oh, a miracle, I received correct USSD answer.

c-e-p-x-u-o, 2017-08-20:
As I understand, the operator above sends USSD in two different formats and you can't set one specific for modem and should detect it dynamically.

oooo1, 2017-08-20:
Is above your real phone number ?
What is your name ?
I will call to you, to discuss the case.
I don' t kniw your dongle model, but I issued this case for my one, I use Huawei E367.

c-e-p-x-u-o, 2017-08-21:
It's fake number, or course.
My name? My phone? To discuss? Who are you? It's github, discuss the case here.
Model number E1550.

oooo1, 2017-08-21:
I am a man who would like to help you due to the same country and may be the same cell carrier.
Ok. Don' t mind.
You can find potential solution for the case at my post in a github space.
At least, itworks with my E367.

c-e-p-x-u-o, 2017-08-21:
As I understand, format is hardcoded for modem model. But it should be detected on the fly.

wdoekes, 2017-08-22:
@marcogaio fixed his USSD issues from #29 by removing "K3765" which he wrongly added to the seven_bit_modems list.

If there are models you want me to add to that list, please specify and I can do so. E367? E1550?

c-e-p-x-u-o, 2017-08-22:
I have only E1550.

oooo1, 2017-08-22:
wdoekes, I advise to add as E1550 as E367.
But I didn' t test with E1550 due to I don' t have it.
With E367 it works fine, at least for me.

wdoekes, 2017-09-26:
Note to self, see also: https://github.com/bg111/asterisk-chan-dongle/issues/231