wdoekes/asterisk-chan-dongle #39: @ after Provider Name

c-e-p-x-u-o, 2017-08-11:
Sometimes, after some time, I see at sign or unprintable square at the end of provider name:

cellgate*CLI> dongle show device state tele2 
-------------- Status -------------
...
  Provider Name           : Tele2�
...

cellgate*CLI> dongle show device state tele2 
-------------- Status -------------
...
  Provider Name           : Tele2@
...

This is expected output:
cellgate*CLI> dongle show device state tele2
-------------- Status -------------
...
  Provider Name           : Tele2
...

c-e-p-x-u-o, 2017-08-11:
sometimes dongle cmd tele2 reset helps
sometimes I need to reboot

wdoekes, 2017-08-22:
An @ sign would be an incorrectly decoded NUL.

But the fact that the output changes worries me more.

If you find any more info, please update this ticket.

c-e-p-x-u-o, 2017-08-22:
And now it's "TELE2" in uppercase:

dongle show device state tele2
...
Provider Name           : TELE2

c-e-p-x-u-o, 2017-08-22:
I don't know, how I do this. I just toggle modem power on the usb hub.

c-e-p-x-u-o, 2017-08-22:
And some time after it become Tele2� again.

c-e-p-x-u-o, 2017-08-22:
Looks like it's not the asterisk-chan-dongle bug!

Welcome to minicom 2.7.1

AT+COPS?
+COPS: 0,2,"25020",2

OK

AT+COPS=?
+COPS: (2,"TELE2 RU","TELE2 RU","25020",2),(1,"MTS-RUS","MTS","25001",2),(1,"Beeline","Beeline","25099",2),(1,"MegaFon)

OK
AT+COPS=?
+COPS: (2,"TELE2 RU","TELE2 RU","25020",2),(1,"MegaFon RUS","MegaFon","25002",2),(1,"Beeline","Beeline","25099",2),(1,)

OK
AT+COPS=?
+COPS: (2,"Tele2@","Tele2�","25020",2),(1,"MTS-RUS","MTS","25001",2),(1,"Beeline","Beeline","25099",2),(1,"MegaFon RUS)

OK

c-e-p-x-u-o, 2017-08-22:
May it be a simcard bug?

c-e-p-x-u-o, 2017-08-22:
Another dongle shows the same error, but all my dongles are E1550 with the same firmware.

c-e-p-x-u-o, 2017-08-22:
I've replaced the simcard, but nothing changed.
Sometimes it shows
+COPS: (2,"Tele2@","Tele2�","25020",2)...
sometimes it shows
+COPS: (2,"Tele2�","Tele2 ","25020",2)...

wdoekes, 2019-11-19:
```diff
--- a/at_parse.c
+++ b/at_parse.c
@@ -103,6 +103,16 @@ EXPORT_DEF char* at_parse_cops (char* str)
                if(marks[3][-1] == '"')
                        marks[3]--;
                marks[3][0] = 0;
+               /* Sometimes there is trailing garbage here;
+                * e.g. "Tele2<U+FFFD>" instead of "Tele2"
+                * see github issue #39 */
+               while (marks[3] > marks[2] && (
+                               (unsigned char)marks[3][-1] < 32 ||
+                               (unsigned char)marks[3][-1] == '@' ||
+                               (unsigned char)marks[3][-1] >= 128)) {
+                       marks[3][0] = 0;
+                       marks[3]--;
+               }
                return marks[2];
        }
 
```

wdoekes, 2019-11-19:
Maybe you want to share the make/model of your dongle as well.

532910, 2019-11-19:
E1550 patched for voice support,
lsusb shows: `12d1:1001 Huawei Technologies Co., Ltd. E161/E169/E620/E800 HSDPA Modem`

532910, 2019-11-19:
The patch fixes both #69 and #39 (this one)

closed 2019-11-20 by wdoekes

532910, 2019-11-20:
Just an acknowledgment that master works fine. Thank you, Walter!

commit 62259b20 (2019-11-20): Trim trailing garbage from provider name causing ast_json_vpack error

The AT+COPS command has been observed to return similar but different
provider names. Sometimes with a trailing illegal character, or a
trailing @.

Dongle make/model E1550 (patched for voice support), identified by Linux
as "Huawei Technologies Co., Ltd. E161/E169/E620/E800 HSDPA Modem".

Example output:

    AT+COPS?
    +COPS: 0,2,"25020",2

    AT+COPS=?
    +COPS: (2,"TELE2 RU","TELE2 RU","25020",2),
           (1,"MTS-RUS","MTS","25001",2),
           (1,"Beeline","Beeline","25099",2),
           (1,"MegaFon)

    AT+ [... cut]