wdoekes/asterisk-chan-dongle #40: Multipart incoming SMSes with Huawei dongle are not concatenated !

CpServiceSpb, 2017-08-20:
There is Asterisk 13.12 at Ubuntu 14.04 LTS no-GUI x64 with Huawei E367 voice enabled.
Two + received messages are "handled" separated +CMGR command (from Asterisk console I can see it) and are not concatenated to one long SMS.
That is message less than 160 symbols (for latin alphabet) or less than 70 symbols (for cyrillic) is received as one, but if is more than 160 or 70 symbols or some messages more than one, all these messages are not concatenated to one long SMS.

Kannel do so. I tested up for 5 parts.

wdoekes, 2017-08-22:
That would be correct. Handling of concatenated messages requires additional work either inside chan_dongle or by the application handling them (you).

Not only it has to keep the messages in a buffer until everything is retrieved, it also has to decide what/when to send when/if a part of the combined message is missing.

I don't have the time nor means to code this up for you, but I'd be willing to review and merge a PR.

oooo1, 2017-08-22:
>Not only it has to keep the messages in a buffer until everything is retrieved, it also has to decide >what/when to send when/if a part of the combined message is missing.
Criterion as I understand, two:
- sender;
- analyze total sms quantity and sms order part # fields of PDU - 3 of 6 for example.
And look at kannel and playsms. It handles multipart sms correctly.

I think chan_dongle has to analize incoming SMS, if it is multipart, extract appropriate message fields as sender # and PDU fields as total sms and sms # in the chain and concatenate it.
And after that only is to move it to a dial plan otherwise it willl be as now - each part calls each sms context in a dial plan.
I think as you know chan_dongle well you will do it mor effectivelly and qcuicker then other folks.

wdoekes, 2017-08-23:
> I think as you know chan_dongle well you will do it [more effectively] and [quicker than] other folks.

Possibly; although I'm not as intimite with all parts of the code as you may think. I just forked it and made a few minor edits.

But since I use it the least of everyone here (I don't have a dongle), it would be the *least useful to me* to create such an addition :)

laigor, 2017-08-23:
I suggest  to set the channel variable indicating the number of the SMS part and the number of parts and then concatenate the SMS parts in the dialplan. This is easier to implement. Less changes to the module code.

mcdull, 2017-08-25:
The SMS part is actually in the part of ${CMGR}.  But I am not sure if it is in the exact location everytime.
For example.
`+CMGR: 0,,40
0891683108707515F14005A10180F600087180529011642318050003710303300230104E2D56FD79FB52A83011000D000A
`
The message part is located at the second line position 59-62 = 0303
03 (Total message) - 03 (current message)

CpServiceSpb, 2017-09-11:
Usefull (may be) link: [Send and Receive Concatenated SMS in PDU mode](https://shekharbiradar.wordpress.com/send-and-receive-concatenated-sms-in-pdu-mode/
)

rodriguescr, 2017-09-23:
This could help you maybe. https://github.com/asiacny/chan-dongle-extended

garronej, 2018-03-30:
Do not try to use chan-dongle-extended. I decided to make the repo private untile I came up with a stable version. The repo of asiacny is a fork of my work that will not run. 

Regards

kalon33, 2019-01-11:
@garronej any updates on this? Is this currently usable? how?

garronej, 2019-01-11:
Yes. It's usable and stable you can use it.
https://garronej.github.io/chan-dongle-extended/

Unfortunately it's not open source just yet. I use it as dependency for a larger project and I am not sure I can make it opensource yes without jeopardize my business.

mcdull, 2019-01-21:
> Yes. It's usable and stable you can use it.
> https://garronej.github.io/chan-dongle-extended/
> 
> Unfortunately it's not open source just yet. I use it as dependency for a larger project and I am not sure I can make it opensource yet without jeopardizing my business.

thank you for your work.  Is this solution work in multiple dongles environment? I have 5 dongles and the traditional code messed up everything by using local file as variable.  And if the sms arrived at the same time from different dongles, the situation even worse.  Sadly it does happen all the time as the carrier will sometimes broadcast the same messages to all SIM.

garronej, 2019-01-21:
Hey @mcdull 
In chan-dongle-extended everything related to SMS management have been entirely rewritten.
It should fix the issues you are currently having and if it does not I would be very interested in your feedback to fix it asap.
Please reach out to me at joseph.garrone.gj@gmail.com I would be happy to help you deploy chan-dongle-extended in your environment.

kalon33, 2019-01-24:
@garronej is this backward compatible with asterisk-chan-dongle regarding config files?

garronej, 2019-01-24:
> @garronej is this backward compatible with asterisk-chan-dongle regarding config files?

Well ``chan-dongle-extended`` manage the file ``dongle.conf``` in realtime. 
When a modem is connected to the host it will automatically add a
new section to the config file. 
You can however change the ```general`` and ``default`` section if you need to 
change let's say ``context`` or ``dtmf``.
If there is already a ``dongle.conf`` file when ``chan-dongle-extended`` is installed 
all the parameters that does not interfer with ``chan-dongle-extended`` will be preserved.

closed 2020-02-19 by wdoekes

CpServiceSpb, 2020-03-12:
I downloaded and built git version on March, 09.
Long sms consised of 3 parts, all in Rusian letters was received successfully.
But when I sent sms even from 1 English letter, I got the folloein error:

[Mar 10 13:03:56] ERROR[3400]: json.c:607 ast_json_vpack: Error building JSON from '{s: s, s: s}': Invalid UTF-8 string.
[Mar 10 13:03:56] ERROR[3400]: Got 11 backtrace records
0: /usr/sbin/asterisk(ast_json_vpack+0xb0) [0x55ac0e136690]
1: /usr/sbin/asterisk(ast_json_pack+0xa5) [0x55ac0e136745]
2: /usr/sbin/asterisk(ast_channel_publish_varset+0x2c) [0x55ac0e24394c]
3: /usr/sbin/asterisk(pbx_builtin_setvar_helper+0x217) [0x55ac0e1af597]
4: /usr/lib/asterisk/modules/chan_dongle.so(+0x2e498) [0x7fb85c155498]
5: /usr/lib/asterisk/modules/chan_dongle.so(+0x21e89) [0x7fb85c148e89]
6: /usr/lib/asterisk/modules/chan_dongle.so(+0x227d2) [0x7fb85c1497d2]
7: /usr/lib/asterisk/modules/chan_dongle.so(+0x2890d) [0x7fb85c14f90d]
8: /usr/sbin/asterisk(+0x599634) [0x55ac0e29b634]
9: /lib/x86_64-linux-gnu/libpthread.so.0(+0x76db) [0x7fb8622c16db]

#10: /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7fb8617fa88f]

Additional test show that 
The same error is also appeared while Sms from 1 part with Russian letters are received.
And there is the old received Sms text in the buffer just after new one.
I think the buffer is not cleaned.

PR #75 (2020-02-19): Concatenated SMS support; Complete locking and single shift tables
Resolves #40