wdoekes/asterisk-chan-dongle #44: Broken audio SIP->GSM

Infactum, 2017-09-13:
Sadly issue  openwrt/telephony#7 is back.
Patch https://github.com/wdoekes/asterisk-chan-dongle/commit/53e83a4f3cafed49f7562cb341cc8f117bbb92b0 by @micmac1 is not working anymore for some reason.

Test system:
Netgear WNDR3800
Huawei E1550 dongle
LEDE Reboot (17.01.2, r3435-65eec8bd5f)
Asterisk 13.9.1

Audio from asterisk to GSM is loud and noisy. Something similar to voice can be heard with about -300 rxgain in dongle.conf
Same dongle works fine on X86 LEDE system (same version).

wdoekes, 2017-09-13:
Okay. It just merged it, I cannot vouch for its validity.

I can revert if you want.

Could it be that it conflicts with this patch?

https://github.com/wdoekes/asterisk-chan-dongle/commit/ab939532c71b991de2f66582d7ee2c553b6a918e

Infactum, 2017-09-13:
Tried to compile without https://github.com/wdoekes/asterisk-chan-dongle/commit/ab939532c71b991de2f66582d7ee2c553b6a918e and it worked.
Unfortunately I can't make proper push request, since reverting https://github.com/wdoekes/asterisk-chan-dongle/commit/ab939532c71b991de2f66582d7ee2c553b6a918e will break package on other systems.

wdoekes, 2017-09-13:
> Unfortunately I can't make proper pull request, since reverting ab93953 will break package on other systems.

What do you mean? I have no system to test against. If you have a big endian one, and it works, then we should go with that. The fix recently added was originally made against a 2015 build, which recieved the "sparc loude noise fix" in 2016. So one of the two fixes likely has to go.

Relevant commits:
```
commit 020ca572109509e0b7f8b1798f3f2ce606394f26
Author: bg_one@mail.ru <bg111@users.noreply.github.com>
Date:   Fri May 27 22:18:22 2011 +0000

    added be platforms

diff --git a/channel.c b/channel.c
index c7a174b..c293c6a 100644
--- a/channel.c
+++ b/channel.c
@@ -651,6 +651,7 @@ static struct ast_frame* channel_read (struct ast_channel* channel)
 
                cpvt->a_read_frame.samples      = res / 2;
                cpvt->a_read_frame.datalen      = res;
+               ast_frame_byteswap_le (&cpvt->a_read_frame);
 /*
                cpvt->a_read_frame.ts;
                cpvt->a_read_frame.len;
@@ -855,6 +856,7 @@ static int channel_write (struct ast_channel* channel, struct ast_frame* f)
                        int iovcnt;
                        struct iovec iov[2];
 
+                       ast_frame_byteswap_le (f);
                        iov[0].iov_base = f->data.ptr;
                        iov[0].iov_len = FRAME_SIZE;
```

```
commit ab939532c71b991de2f66582d7ee2c553b6a918e
Author: bg111 <bg111@users.noreply.github.com>
Date:   Sun Sep 25 12:38:12 2016 +0800

    fix Sparc loud noise problem #201

diff --git a/channel.c b/channel.c
index 0c96fa4..f695d99 100644
--- a/channel.c
+++ b/channel.c
@@ -820,6 +820,7 @@ static int channel_write (struct ast_channel* channel, struct ast_frame* f)
                        }
                }
 
+               ast_frame_byteswap_le(f);
 
                if (pvt->a_timer)
                {
```

And now 53e83a4f3cafed49f7562cb341cc8f117bbb92b0 from openwrt.

There are now too many swaps for sure:

```
diff --git a/channel.c b/channel.c
index e689e9f..4e84bc4 100644
--- a/channel.c
+++ b/channel.c
@@ -576,7 +576,7 @@ static void timing_write(struct pvt* pvt)
                        iovcnt = mixb_read_n_iov (&pvt->a_write_mixb, iov, FRAME_SIZE);
                        mixb_read_n_iov (&pvt->a_write_mixb, iov, FRAME_SIZE);
                        mixb_read_upd (&pvt->a_write_mixb, FRAME_SIZE);
-                       change_audio_endianness_to_le(iov, iovcnt);
+                       change_audio_endianness_to_le(iov, iovcnt); // XXX
                }
                else if (used > 0)
                {
@@ -590,7 +590,7 @@ static void timing_write(struct pvt* pvt)
                        iov[iovcnt].iov_base    = silence_frame;
                        iov[iovcnt].iov_len     = FRAME_SIZE - used;
                        iovcnt++;
-                       change_audio_endianness_to_le(iov, iovcnt);
+                       change_audio_endianness_to_le(iov, iovcnt); // XXX
                }
                else
                {
@@ -679,7 +679,7 @@ static struct ast_frame* channel_read (struct ast_channel* channel)
        if (pvt->a_timer && ast_channel_fdno(channel) == 1)
        {
                ast_timer_ack (pvt->a_timer, 1);
-               timing_write (pvt);
+               timing_write (pvt); // XXX
                ast_debug (7, "[%s] *** timing ***\n", PVT_ID(pvt));
        }
 
@@ -727,7 +727,7 @@ static struct ast_frame* channel_read (struct ast_channel* channel)
 
                cpvt->a_read_frame.samples      = res / 2;
                cpvt->a_read_frame.datalen      = res;
-               ast_frame_byteswap_le (&cpvt->a_read_frame);
+               ast_frame_byteswap_le (&cpvt->a_read_frame); // XXX
 /*
                cpvt->a_read_frame.ts;
                cpvt->a_read_frame.len;
@@ -910,7 +910,7 @@ static int channel_write (struct ast_channel* channel, struct ast_frame* f)
                        }
                }
 
-               ast_frame_byteswap_le(f);
+               ast_frame_byteswap_le(f); // XXX (added in ab939532c7)
 
                if (pvt->a_timer)
                {
@@ -945,7 +945,7 @@ static int channel_write (struct ast_channel* channel, struct ast_frame* f)
                        int iovcnt;
                        struct iovec iov[2];
 
-                       ast_frame_byteswap_le (f);
+                       ast_frame_byteswap_le (f); // XXX
                        iov[0].iov_base = f->data.ptr;
                        iov[0].iov_len = FRAME_SIZE;
 
```

micmac1, 2017-09-13:
Hi all,

How about putting out the openwrt patch. I'd then send a PR to openwrt to get the source in sync with your master. If the endianess issue apparently was already fixed earlier I apologize for the regression.

Regards,
Sebastian

Am 13. September 2017 14:24:11 MESZ schrieb Walter Doekes <notifications@github.com>:
> ab93953 will break package on other systems.
>
> What do you mean? I have no system to test against. If you have a big
> endian one, and it works, then we should go with that. The fix recently
> added was [...]

wdoekes, 2017-09-13:
There. Reverted. Let me no if additional fixage is needed.

closed 2017-09-13 by wdoekes

Infactum, 2017-09-13:
Doen't work in my setup.
Reverting https://github.com/wdoekes/asterisk-chan-dongle/commit/ab939532c71b991de2f66582d7ee2c553b6a918e was fine.

wdoekes, 2017-09-13:
Hah, too fast. Okay. Hang on while I rewrite history.

commit 905954fe (2017-09-13): Revert "Fix audio endianess problem on big endian systems."

This reverts commit 53e83a4f3cafed49f7562cb341cc8f117bbb92b0.

Likely that commit conflicted with ab939532c71b991de2f66582d7ee2c553b6a918e.

Closes #44.
commit 4ef5ad7e (2017-09-13): Revert "fix Sparc loud noise problem #201"

This reverts commit ab939532c71b991de2f66582d7ee2c553b6a918e.

Apparently there was too much byteswapping (on big-endian machines)
going on with both ab93953 and 53e83a4. Reverting ab93953 appears to
work for @Infactum.

Closes #44.