wdoekes/asterisk-chan-dongle #46: Support 4G/LTE Dongles

greencall, 2017-09-26:
Hi,
We are using successfully chan-dongle with 3G modems but in this moment we need to upgrade to 4G modems.
Is there any chance to add some 4G modems in the library? 
Or maybe is better we choose a model and send you all specifications?

Thanks

CpServiceSpb, 2017-11-05:
As I know noone 4G dongle supports voice.

CpServiceSpb, 2020-04-18:
I was wrong. As I have known some time ago, there is Huawei E352 supporting voice.
But only one of two submodel of E352**B**.
Exact with B letter.
Firmware : 21.158.23.00.209

CpServiceSpb, 2020-04-18:
> 
> 
> I was wrong. As I have known some time ago, there is Huawei E352 supporting voice.
> But only one of two submodel of E352**B**.
> Exact with B letter.
> Firmware : 21.158.23.00.209

Or Huawei E352 is not 4G but 3G dongle.