wdoekes/asterisk-chan-dongle #48: Time to time dongle disconnection from Asterisk chan_dongle.

oooo1, 2017-10-22:
Asterisk 13.12 with Huawei E367 and the last this chan_dongle version is installed at Ubuntu 14.04 x64.

Time to time, Asterisk chan_dongle lose 3G dongle connection.
That is 'dongle show devices' sows 'Not connected' .

The worst thing in the situation that there are not any useful messages in Asterisk messages log.
Only:
[Oct 22 12:53:50] ERROR[11786] chan_dongle.c: [Huawei7] Lost connection to Dongle

Sometimes such message is appeared:
[Oct 20 08:55:45] WARNING[8356][C-0000001b] channel.c: [Dongle/HuaweiTl25667-0100000017] Don't know how to indicate condition 22
[Oct 20 08:55:52] ERROR[11786] at_response.c: [HuaweiTl25667] Receive NO CARRIER
In the case, after this message dongle continued working.

Firstly, I thought that the reason is autosuspend - no any incoming / outgoing calls were made within 12 hours.
But **ls /dev** shows all devices: audio, data and someelse device for this dongle.

Sendind sms from minicom was successfull.
That is minicom -D /dev/dongle_data and 
AT+CMGF=1
AT+CMGS="+..................."
Test. 
<ctrl+z>
passed successfully.

Helps only Asterisk reload command.
What and where are reasons of sudden disconnection at ?

gamba47, 2017-10-23:
Some times this is a power issue, please connect your dongle directly to your motherboard. Avoid use of usb hubs at least for this test.

First of all try to detect on "dmesg" if you see any disconection or changes on devs
Take a look at this https://github.com/wdoekes/asterisk-chan-dongle/issues/37

Best regards.

gamba47

oooo1, 2017-10-29:
I thought sp.
But I have 7-port USB haub with connected external power supply.
And it was happended after some non using time.
I thought that the reason is autosuspend.
But as I wrote modem is not connected in Asterisk only (chan_dongle) , in the time I can send sms, for example, using minicom.

oooo1, 2017-11-11:
I set up power/control="on" and power/autosuspend="-1" for port occupied by dongle (such as 2-1.1.1)
Reboot the server.
I have 4.2.0-42-generic kernel.
And after about 17 hours I got "dongle lost connection" at log.
And dongle show device state MySuperDongle gave out the following:
```
 Device                  : MySuperDongle 
  State                   : Not connected Start scheduled
  Audio                   : /dev/ModemAudioPort
  Data                    : /dev/ModemDataPort
  Voice                   : No
  SMS                     : No
  Manufacturer            :
  Model                   :
  Firmware                :
  IMEI                    :
  IMSI                    :
  GSM Registration Status : Unknown
  RSSI                    : 0, <= -113 dBm
  Mode                    : No Service
  Submode                 : No service
  Provider Name           : NONE
  Location area code      :
  Cell ID                 :
  Subscriber Number       : Unknown
  SMS Service Center      :
  Use UCS-2 encoding      : No
  USSD use 7 bit encoding : No
  USSD use UCS-2 decoding : Yes
  Tasks in queue          : 0
  Commands in queue       : 0
  Call Waiting            : Disabled
  Current device state    : stop
  Desired device state    : start
  When change state       : now
  Calls/Channels          : 0
    Active                : 0
    Held                  : 0
    Dialing               : 0
    Alerting              : 0
    Incoming              : 0
    Waiting               : 0
    Releasing             : 0
    Initializing          : 0
```
And nothing is up.

wdoekes, 2018-03-05:
I'm sorry, this is not something I can debug. Feel free to discuss this issue further here. If you come up with more clues I may be able to fix something for you.

CpServiceSpb, 2020-04-18:
So, discovery dongle by imei and imsi don; t work on CubieTruck (armv7l architecture) .
Ports must be specified exactly.