wdoekes/asterisk-chan-dongle #49: make install error on centOS 7 x6/usr/bin/install: missing destination file operand after ‘chan_dongle.so’

alaa2003, 2017-12-01:
[root@freepbx asterisk-chan-dongle]# make install
strip chan_dongle.so
/usr/bin/install -c -m 755 chan_dongle.so
/usr/bin/install: missing destination file operand after ‘chan_dongle.so’
Try '/usr/bin/install --help' for more information.
make: *** [install] Error 1

wdoekes, 2017-12-03:
The first three commands are:
```
$ ./bootstrap
$ ./configure --with-astversion=13.7
$ make
```
Those are the *only* commands.

The rest of the README says:

> If you run a different version of Asterisk, you'll need to update the 13.7 as appropriate, obviously.
> If you did not `make install` Asterisk in the usual location and configure cannot find the asterisk header files in /usr/include/asterisk, you may optionally pass --with-asterisk=PATH/TO/INCLUDE.

It does not say "type `make install`". 

Right now, the build script, has no module directory detection. It probably should, so make install works for those (few) that use it. For now, you'll have to find the moduledir yourself. It's probably in `/usr/lib/asterisk/modules`. Just copy the `chan_dongle.so` there.

alaa2003, 2017-12-04:
Thanks a lot it worked..
as you mentioned copy the chan_dongle.so to :
x32 OS module dir mostly /usr/lib/asterisk/modules
x64 OS module dir mostly /usr/lib64/asterisk/modules
#then load it from asterisk
asterisk -rvvvvv
module load chan_dongle.so

wdoekes, 2018-03-05:
Should work for you now.

commit cfab716b (2018-03-05): Improve asterisk/modules dir autodetection

Previously ./configure would continue after not finding DESTDIR for the
module and then fail at 'make install'. Now it requires you to set
DESTDIR if autodetection fails. Also added lib64 in the default paths to
try.

Reported-by: @alaa2003
Closes: #49
closed 2018-03-05 by wdoekes