wdoekes/asterisk-chan-dongle #52: Any action on incoming call

essenemari, 2018-03-09:
Hi,
@wdoekes, first of all thanks a lot for your work on this asterisk-chan-dongle distribution.

Anyone,
I am looking for some help, so probably the best label will "help wanted". So, if someone would able to give me some hints (really basic level - just to get up), i would really appreciate it.

**Current conf (test environment):**
Ubuntu 17.10
Asterisk 15.2.2
E173u-2, 11.126.85.02.52, voice enabled, unlocked.

I believe, I have correctly configured asterisk with chan_dongle because I can send and receive SMS through dongle.

**Issue:**
But I have no idea, how manage (or just test!) incoming call. I have **always** "busy signal"  when trying to call to my dongle... Tried many ways with extension.conf. I did copy the file to /etc/asterisk/ folder and put there many testing data to see what happen. Nothing worked. Let's take this (example):

/etc/asterisk/extensions.conf
```
[general]
[default] //as default context defined in /etc/astersk/dognle.conf
exten => s,n,DongleSendSMS(dongle0,+00123456789,"Hello how are you, Danila?",1440,yes)
;exten => s,1,Wait(6)
```
My understanding is that dongle should recognize any call and send the SMS to +00123456789. But when calling then always I hear "User busy". Tried also to play with Wait() function but still "User busy".
Tested a lot of other variants but no progress :| Tried even to remove extensions.conf from everywhere to see if maybe i will see some useful EM while asterisk reloading, but no... all seems to be working normally and then if calling to dongle ... dongle is busy... /var/log/asterisk empty... 

Please help me understand how to move it forward, for example how send SMS to +00123456789 if someone is calling to dongle or anything else that could help me to see if I can play with incoming calls. 

Any advise will be helpful.

Thanks!
..................................
My long term goal: by having MSISDN  + HUAWEI E173u-2 + RPI + asterisk + asterisk-chan-dongle trying to simple create IVR  to perform two actions e.g. if someone will call to +ccxxxxxxxxx,2;1 then action#1 (CURL smth to server), if +ccxxxxxxxxx,2;2 then action#2 CURL smth else to server).

standardgbg, 2018-03-11:
Hi

I had this problem as well. There is most certainly something wrong with you config.
Start by logging in to asteriskCLI with "asterisk -r" and up the logging lever with "core set verbose 3"
Call the number and start scratching your head :)

essenemari, 2018-03-11:
@standardgbg 
MANY thanks for your response. You've helped me a lot because I realized that FreePBX installation seems to be obligatory. In fact tried avoid FreePBX installation because it will run on RPI - you know, less things on RPI equals SD card longer life. I thought that it asterisk is able to handle that on its own but based on my research and also your confirmation seems it will not work or is too hard for me to set  up.
once again - thank you! I will work on this upcoming weeknights :)

standardgbg, 2018-03-12:
No problem.

Regarding your problem with no loggs.
I think your dongle.conf must be wrong.

The only thing you need to change is 
exten=+46XXXXX (Your phone number) 

and check in dmesg that your modem has switched to the following ports and is now in modem-mode and not storage mode (and can handle voice, best checked in the PC program "DC unlocker")
audio=/dev/ttyUSB1              ; tty port for audio connection;        no default value
data=/dev/ttyUSB2               ; tty port for AT commands;             no default value

And I take it that you've installed FreePBX and setup a trunk and incoming route.

wdoekes, 2018-03-12:
(1) You don't need FreePBX for functionality ever. You can use FreePBX for convenience, but it will not do things that you cannot do without it.
(2) You do need logging. See `/etc/asterisk/logger.conf` and specifically the `console=>` line if you're not seeing any verbose logs. (Don't forget to restart (or logger reload) after change.)

essenemari, 2018-03-12:
thank You both for your valuable help. 

- It turned out that problem was with dongle itself means despite the dc-unlocker showed voice enabled, the same was for "dongle show device state dongle0" it was not possible to call from/to dongle through Mobile Partner on Windows. 

- I decided to change firmware.

- Now I can call/and receive call through Media Partner on Windows. On the Asterisk I can finally see debugging and errors which is extremely needed to further proceed. Lesson for me from this case is that if Voice is disabled (always empirically test! - DC-Unlocker showed fake) there is no logs and no debugging while call attempt. 

Once again! Many thanks for you, gentlemen!

essenemari, 2018-03-13:
Gentleman, 
Problem has been fixed! Thank You for your help!

Issue description:
- I bought dongle Huawei e173u-2 from local stock.
- All tools gave me clear information that Voice feature is enabled (more info above), so initial requirement to play with dongle was fulfilled.
- However the issue was, that no matter what configuration on asterisk I had (extension.conf/dongle.conf), I always had busy line signal plus there was no logs while call attempt on Asterisk.
- Because of no logs, I assumed that my asterisk/dongle configuration is broken however I was not able to find where its broken.
- Finally, in desperation, I have posted my issue and someone advised me to empirically check if I am really able to call from Mobile Partner program on Windows.
- Initially I was surprised because some versions of Media Partner don’t have “call option”, so just assumed that the call /out and in/ can be done just with other software but not Mobile Partner.
- someone form Asterisk community presented the version with Mobile Partner with such feature.
- It turned out that line (on Windows at this time) is also busy.
- Tried to play with unlocking dongle (VOICE) by AT commands, but all commands that I found showed “COMMAND NOT SUPPORT”.
- Found firmware: 11.126.85.00.209 for educational purposes. Flashed device and I was able to make/receive call on Media Partner.
- Tested Asterisk: I was able to create “Hello world” configuration.

Problem fixed! :+1:

closed 2018-03-13 by essenemari