wdoekes/asterisk-chan-dongle #53: Unhandled PID value, only SMS supported (Name as sender number)

standardgbg, 2018-03-11:
Hi!

First off I want to **thank you** for making this patched version of chan_dongle. 
Its working great and I love it!
I've just found one problem with it and this is goes for all of the chan_dongle versions.

I have a E1750 setup to forward all incoming SMS with mail.
But it can't handle when the sender uses a alphabetical sender name instead of a number.

`[2018-03-10 18:32:02] WARNING[27972]: at_response.c:1232 at_response_cmgr: [dongle0] Error parsing incoming message '+CMGR: 0,,124
**I HAVE REMOVED MY SMS PID_VALUE'** at position 53: Unhandled PID value, only SMS supported`

There is similar bugs reported in the other versions of chan_dongle and some patch files are available but I can't get them to work in the version.
Any help would be appreciated :)

wdoekes, 2018-03-12:
I don't see you PID value, so I cannot comment on what it is, but looking on the internet I see for example this:
https://sourceforge.net/p/raspbx/discussion/bugreports/thread/55bc0a67/
where the pid value is 0x39 == 57

Which would be TP-PID 57 "SC-specific; usage based on mutual agreement between the SME and the SC"

https://en.m.wikipedia.org/wiki/GSM_03.40#Protocol_Identifier

SME = Short Messaging Entity
SC = Service Center

I suppose allowing anything below 128 would work without any changes.

Try this (compile tested only):
```
diff --git a/pdu.c b/pdu.c
index 9db8a48..55cbb13 100644
--- a/pdu.c
+++ b/pdu.c
@@ -761,9 +761,14 @@ EXPORT_DEF const char * pdu_parse(char ** pdu, size_t tpdu_length, char * oa, si
 		return "Can't parse PID";
 	}
 
-	/* TODO: support other types of messages */
 	if (pid != PDU_PID_SMS && !(0x41 <= pid && pid <= 0x47) /* PDU_PID_SMS_REPLACE_MASK */) {
-		return "Unhandled PID value, only SMS supported";
+		if (pid < 128) {
+			/* See: https://en.m.wikipedia.org/wiki/GSM_03.40#Protocol_Identifier
+			 * Not sure how to handle these, allow for now.. */
+			ast_debug(1, "Allowing TP-PID value 0x%hhx\n", (unsigned char)pid);
+		} else {
+			return "Unhandled PID value, only SMS supported";
+		}
 	}
 
 	dcs = pdu_parse_byte(pdu, &pdu_length);
```

standardgbg, 2018-03-12:
Hi,

Sorry, but I removed the PID data for privacy reasons. 
If i run it through a PDU decoder I get TP-PID: 39.
I will test and report back :)
Thanks!

wdoekes, 2018-03-12:
Glad to help.

> It doesn't decode the sender alphabetical name

That is a different problem. It's probably located somewhere near the parsing of `TP-OA`.

I'd still need (parts of) your censored data to try and fix that.

standardgbg, 2018-03-12:
I'll PM it to you :)

standardgbg, 2018-03-12:
Correction:
It's all working great!
The sender I used to test an alphabetical sender name started using a phone number for some reason. I tested another sender and its all working like a charm!
I'm happy. Again, thank you! 🥇

commit 6ba9cf26 (2018-03-12): pdu/parsing: Allow all values for TP-PID

Closes #53, reported by @standardgbg.

See: http://www.3gpp.org/ftp//Specs/archive/23_series/23.040/23040-e00.zip
closed 2018-03-12 by wdoekes