wdoekes/asterisk-chan-dongle #54: help with compiling issues

markosjal, 2018-05-05:
i have tried building this chan_dongle and get errors in the build process

at 

automake -a , I get

automake: error: no 'Makefile.am' found for any configure output

am I doing something wrong?

I am building on Ubuntu 16.04  32 bit

THanks

wdoekes, 2018-05-07:
```$ cat bootstrap 
#!/bin/sh

aclocal
autoconf

# We don't use Makefile.am, but we *do* use automake to add
# "missing" files used by configure.
automake -a 2>&1 | grep -vF Makefile.am; true
```

markosjal, 2018-05-13:
Is there a "how to" for building? I see this question all over the internet and although you may understand what you wrote, I have no idea how/where to implement this.