wdoekes/asterisk-chan-dongle #55: Dialing from console

sharun-s, 2018-06-08:
This is more of a question that an issue. I am a new user to asterisk so a bit confused about the addressing used in the extensions.conf while making outgoing calls. 

I have a very basic setup(no sip/e1/t1 etc) and am only trying to use the dongle for sms send/recv and making and receiving voice calls. I have managed to get the sms and receiving voice calls to work. But I am finding it hard to figure out from the docs/internet how to dial a number from the console.
Is this possible?

I guess I have to use  
channel originate <tech/data> extension [exten@][context]

but I can't figure out what these parameters are and what context I need to set. 

I also see an alternate "console dial some-phn-number" in certain examples but I get console module not loaded errors. So could someone please provide a simple example or the link to documentation I can look up. Thanks!

sharun-s, 2018-06-12:
Turns out chan_console.so and chan_alsa.so weren't being built during my install on Ubuntu 16.04. Both were dependent on (missing) pulseaudio libraries. 
Simplest way to fix was to run the contrib/scripts/install_prereq script included with the asterisk(15.4) source, followed by configure and make and then moving the newly built libraries to /usr/lib/asterisk/modules. Followed by changing the module.conf to use chan_alsa instead of chan_oss.so. So closing this issue.

closed 2018-06-12 by sharun-s

wdoekes, 2018-06-13:
Thanks for the update!