wdoekes/asterisk-chan-dongle #57: Fail to decode SMS message PID type 39.

marcogaio, 2018-06-28:
`Jun 28 11:21:15 eraldo asterisk[30997]: WARNING[31063]: at_response.c:1230 in at_response_cmgr: [dongle0] Error parsing incoming message '+CMGR: 0,,147#015#01207919323505952000407D0D7B49B0C39008160821112318094D17A794E0F83A6C926683E0E93E5E11368067BC16E2F192C8673819CE53468063B9F417079795C2697DDF434881E669741E4303D0C82BFE9F2701A347FB3DF20797A5CB697E565D0189D0EB7C3F432A84C064D9B5317489A1E87E5E97118042FCB41E3B79B9E76D7C3F232284C06D5E76179199406B9DF73BA3C0D9A97E5F6B43E0D' at position 47: Unhandled PID value, only SMS supported`

The PDU is:  `07919323505952000407D0D7B49B0C39008160821112318094D17A794E0F83A6C926683E0E93E5E11368067BC16E2F192C8673819CE53468063B9F417079795C2697DDF434881E669741E4303D0C82BFE9F2701A347FB3DF20797A5CB697E565D0189D0EB7C3F432A84C064D9B5317489A1E87E5E97118042FCB41E3B79B9E76D7C3F232284C06D5E76179199406B9DF73BA3C0D9A97E5F6B43E0D`
and with online decoder decode flawlessy, is an operator messages that inform me that my SIM is expiring.

Internet say me that PID type 39 in 'videotext', and this is really strange...

Thanks.

wdoekes, 2018-06-28:
> ... 07D0D7B49B0C39 ...
> Internet say me that PID type 39 in 'videotext', and this is really strange...

No, it's 0x39 == 57, which is "SC-specific; usage based on mutual agreement between the SME and the SC"

marcogaio, 2018-06-29:
Ops! I've not considered was an hexadecimal number! Sorry!

There's something i can do to support that type of messages?

pali, 2019-07-20:
Seems that more online PDU decoders have no problem to decode such SMS... See (open and put there that PDU):
https://smspdu.benjaminerhart.com/
https://www.diafaan.com/sms-tutorials/gsm-modem-tutorial/online-sms-pdu-decoder/

wdoekes, 2019-07-22:
Okay. My focus is not on chan-dongle development; so I probably won't implement any of that.

Pull requests are welcome though ;-)

wdoekes, 2020-05-03:
Closing. Nothing here I'll be implementing without help.

closed 2020-05-03 by wdoekes

pali, 2020-05-03:
@wdoekes What kind of help you need?

I have already sent you two links for decoding such PDU.
It is SMS of type "SMS-DELIVER", PDU type is Telematic interworking (Type: SC specific value).
Text of that SMS is:
```
Questa SIM scadra' 30/07/2018. Nei 30gg precedenti tale data potrai solo ricevere chiamate ed SMS. Ricarica per continuare ad usare i nostri servizi
```

First link also contains source code of that decoder, relevant parts are here:
https://github.com/tladesignz/jsPduDecoder/blob/master/source/jspdudecoder.js#L494-L496
https://github.com/tladesignz/jsPduDecoder/blob/master/source/jspdudecoder.js#L588-L644

wdoekes, 2020-05-03:
> I have already sent you two links for decoding such PDU.

Well, I guess that means it's done then.

Sarcasm aside: what do you want to happen? Should it be handled as if it's just a regular SMS?

> What kind of help you need?

Well, a PR never hurts :)

pali, 2020-05-03:
> Should it be handled as if it's just a regular SMS?

Why not? I would expect that body of that SMS would be handled in the same way like body of other messages...

pali, 2020-06-06:
Just to note, PID is information for SC how it should deliver particular SMS. It does **not** describe format of SMS.

So it is purely incorrect if SMS decoder to plain text is looking and trying to interpret PID.

Format of SMS is defined in DCS field, not in PID field.

Also GSM standard says that MS should interpret reserved or unsupported PID as if it was value zero (which is what most of SMS have).