wdoekes/asterisk-chan-dongle #60: Incoming SMS never recognized

KaitoCross, 2018-08-08:
Hey everyone! I am using a Huawei K3250 (voice works fine with this driver) and tried receiving sms using the example dialplan in the readme. But nothing is showing up in the textfile ever. How can i fix this?

KaitoCross, 2018-08-21:
On top of that, whenever i send SMS, the log starts repeating the message
[Aug 21 03:59:17] WARNING[30186][C-00000001] channel.c: Exceptionally long voice queue length queuing to Local/sms@dawei-00000000;2

magcks, 2020-02-14:
Did you enable smsaspdu?

Could you please post the payload from the +CMGR response? You can insert the following line to `at_response.c#L1210` to print the CMGR to the asterisk console:
`ast_log(LOG_WARNING, "[%s] CMGR: %s\n", PVT_ID(pvt), str);`

grwlf, 2020-10-16:
@magcks  could you please take a look at https://github.com/wdoekes/asterisk-chan-dongle/issues/121  Sounds like a similar issue. What is `smsaspdu`? I can't find an option like this

magcks, 2020-10-23:
Smsaspdu was an option to force the modem to send PDU or text formatted messages. I removed this option as the text mode is very hard to maintain (you need multiple character checks, to determine if this message can actually be sent via text mode).

grwlf, 2020-10-23:
Thanks, got it. I can find this option in some posts, it should be an older versions. But it turns out that my problem is not related to it.