wdoekes/asterisk-chan-dongle #64: GL-AR150 - OpenWrt

adamiD, 2019-02-25:
Hello guys,

I have my system working fine, but sometimes the modem stops communicating, it is necessary to remove and reconnect, but I wanted the task to be automatic. (my device lets you control the power of the USB port).

My question is, any suggestions to trigger a script that does the job?

Greetings everyone!

dentys03, 2019-02-25:
hi,
what does "stops communicating mean"?

have you tried this:
asterisk -rx "dongle cmd dongle0 AT^CFUN=1,1"

?

adamiD, 2019-02-25:
> hi,
> what does "stops communicating mean"?
> 
> have you tried this:
> asterisk -rx "dongle cmd dongle0 AT^CFUN=1,1"
> 
> ?

The device is no longer available on the system, it is necessary to remove it from the USB port and re-insert ... my intention is to monitor this state and the script does this automatically.

dentys03, 2019-02-27:
chan_dongle.c:423 do_monitor_phone: [5] Lost connection to Dongle

Maybe hook here your script?

Please do post your solution when you find it.

closed 2019-04-24 by adamiD