wdoekes/asterisk-chan-dongle #67: how do you install this?

kbtang88, 2019-04-16:
how do you install this

ger-pakon, 2019-04-16:
let's have a look at this file: https://github.com/wdoekes/asterisk-chan-dongle/blob/master/README.md

kbtang88, 2019-04-17:
[root@freepbxhome asterisk-chan-dongle]# ./configure --with-astversion=16.2.1
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking target system type... x86_64-unknown-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for style of include used by make... GNU
checking dependency style of gcc... none
checking how to run the C preprocessor... gcc -E
checking for strip... strip
checking for rm... rm
checking for library containing iconv... none required
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking fcntl.h usability... yes
checking fcntl.h presence... yes
checking for fcntl.h... yes
checking for stdlib.h... (cached) yes
checking for string.h... (cached) yes
checking sys/time.h usability... yes
checking sys/time.h presence... yes
checking for sys/time.h... yes
checking termios.h usability... yes
checking termios.h presence... yes
checking for termios.h... yes
checking whether asterisk.h in ../include... no
checking whether asterisk.h in /usr/include... no
checking whether asterisk.h in /usr/local/include... no
checking whether asterisk.h in /opt/local/include... no
configure: error: Can't find "asterisk.h"

wdoekes, 2019-04-17:
You'll need the asterisk header files.

If you made Asterisk from source, they came with `make install` there.

If you installed Asterisk through a package manager, you should install the corresponding -dev/-devel packages.

wdoekes, 2019-04-17:
(I'm guessing you're after these: https://packages.debian.org/sid/all/asterisk-dev/filelist )

kbtang88, 2019-04-17:
where from?

thanks

kbtang88, 2019-04-17:
the only thing 

[root@freepbxhome ~]# cd /usr/include/asterisk
[root@freepbxhome asterisk]# ls
[root@freepbxhome asterisk]# ls -l
total 0

i have nothing in there

kbtang88, 2019-04-17:
i did it i have reinstall once the upgrade i was running 13 version working on 16.3.0

wdoekes, 2019-08-26:
I assume we can close this now :)

closed 2019-08-26 by wdoekes