wdoekes/asterisk-chan-dongle #69: json.c:607 ast_json_vpack: Error building JSON from '{s: s, s: s}': Invalid UTF-8 string.

532910, 2019-08-26:
asterisk 1:16.2.1~dfsg-1 debian buster

```
[donglename] Got SMS from +phonenumber: 'test'
[2019-08-26 03:47:05] ERROR[23550]: json.c:607 ast_json_vpack: Error building JSON from '{s: s, s: s}': Invalid UTF-8 string.
[2019-08-26 03:47:05] ERROR[23550]:   Got 10 backtrace records
# 0: /usr/sbin/asterisk(ast_json_pack+0x99) [0x564005b91459]
# 1: /usr/sbin/asterisk(ast_channel_publish_varset+0x2b) [0x564005c0c03b]
# 2: /usr/sbin/asterisk(pbx_builtin_setvar_helper+0x149) [0x564005bc6f09]
# 3: /usr/lib/asterisk/modules/chan_dongle.so(+0x1865e) [0x7fe5ed74265e]
# 4: /usr/lib/asterisk/modules/chan_dongle.so(+0xd1b0) [0x7fe5ed7371b0]
# 5: /usr/lib/asterisk/modules/chan_dongle.so(+0xdbd2) [0x7fe5ed737bd2]
# 6: /usr/lib/asterisk/modules/chan_dongle.so(+0x12bbb) [0x7fe5ed73cbbb]
# 7: /usr/sbin/asterisk(+0x18843f) [0x564005c2a43f]
# 8: /lib/x86_64-linux-gnu/libpthread.so.0(+0x7fa3) [0x7fe5f1604fa3]
# 9: /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7fe5f135f4cf]
```

wdoekes, 2019-08-26:
Most likely culprit would be:
```
--- a/channel.c
+++ b/channel.c
@@ -1222,8 +1222,12 @@ static void set_channel_vars(struct pvt* pvt, struct ast_channel* channel)
        //ast_string_field_set (channel, language, CONF_SHARED(pvt, language);
 #endif /* ^11- */
 
-       for(idx = 0; idx < ITEMS_OF(dev_vars); ++idx)
+       for(idx = 0; idx < ITEMS_OF(dev_vars); ++idx) {
+               ast_log(LOG_ERROR, "DEBUG: %s = %s\n",
+                       (dev_vars[idx].name ? dev_vars[idx].name : "(null)"),
+                       (dev_vars[idx].value ? dev_vars[idx].value : "(null)"));
                pbx_builtin_setvar_helper (channel, dev_vars[idx].name, dev_vars[idx].value);
+       }
 }
 
 /* NOTE: called from device and current levels with locked pvt */
```
Can you check?

I assume that one of those values is NULL and that ast_json_pack in ast_channel_publish_varset doesn't like it.

Additionally, try this for more backtrace output:
```
diff --git a/Makefile.in b/Makefile.in
index c515874..be0bc18 100644
--- a/Makefile.in
+++ b/Makefile.in
@@ -38,9 +38,9 @@ INSTALL = @INSTALL@
 CHMOD = chmod
 
 # -DAST_MODULE=\"$(PROJM)\" -D_THREAD_SAFE
-CFLAGS  = @CFLAGS@ -I$(srcdir) -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self \
+CFLAGS  = @CFLAGS@ -g3 -I$(srcdir) -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self \
        @CPPFLAGS@ @DEFS@ @AC_CFLAGS@
-LDFLAGS = @LDFLAGS@ 
+LDFLAGS = @LDFLAGS@ -g3
 SOLINK  = @SOLINK@
 LIBS    = @LIBS@
 DISTNAME= @PACKAGE_TARNAME@-@PACKAGE_VERSION@.r@PACKAGE_REVISION@
@@ -51,7 +51,7 @@ VPATH = @srcdir@
 all: @TARGET@
 
 install: all
-       $(STRIP) $(PROJM)
+       #$(STRIP) $(PROJM)
 ifneq (@DESTDIR@,)
        $(INSTALL) -m 644 $(PROJM) @DESTDIR@
 else
```

532910, 2019-08-26:
Unable to apply the first patch above:
```
% git clone https://github.com/wdoekes/asterisk-chan-dongle asterisk-chan-dongle-test
% cd asterisk-chan-dongle-test 
% cat ../1.diff| patch -p1    
patching file channel.c
Hunk #1 FAILED at 1222.
1 out of 1 hunk FAILED -- saving rejects to file channel.c.rej
```

532910, 2019-08-26:
And the second one too:
```
% cat ../2.diff| patch -p1   
patching file Makefile.in
Hunk #1 FAILED at 38.
Hunk #2 FAILED at 51.
2 out of 2 hunks FAILED -- saving rejects to file Makefile.in.rej
```

532910, 2019-08-31:
@wdoekes could you check your patches. I can't apply them.

wdoekes, 2019-09-03:
If you cannot get those patches to work, I don't think we can help each other.

532910, 2019-09-03:
Could you attach them as files, my be I can't correctly copy-paste them?

532910, 2019-11-19:
Walter, could you please check the patches, I can't apply them to the master. May be I can't copy them correct. Could you recheck them and attach as files.

Or could you create a separate branch that I can checkout?

wdoekes, 2019-11-19:
Try the patch `--ignore-whitespace` option. And make sure the Makefile(.in) gets tabs, not spaces.

532910, 2019-11-19:
1. `--ignore-whitespace` solves the patch problem
2. I see the same call trace with both patches applied:
```
 [donglename] Got SMS from +phonenumber: 'test '
[2019-11-19 19:29:34] ERROR[2847]: channel.c:1226 set_channel_vars: DEBUG: DONGLENAME = donglename
[2019-11-19 19:29:34] ERROR[2847]: channel.c:1226 set_channel_vars: DEBUG: DONGLEPROVIDER = donglename�
[2019-11-19 19:29:34] ERROR[2847]: json.c:607 ast_json_vpack: Error building JSON from '{s: s, s: s}': Invalid UTF-8 string.
[2019-11-19 19:29:34] ERROR[2847]:   Got 11 backtrace records
# 0: /usr/sbin/asterisk(ast_json_pack+0x99) [0x55e07cae1449]
# 1: /usr/sbin/asterisk(ast_channel_publish_varset+0x2b) [0x55e07cb5c01b]
# 2: /usr/sbin/asterisk(pbx_builtin_setvar_helper+0x149) [0x55e07cb16f79]
# 3: /usr/lib/asterisk/modules/chan_dongle.so(+0x14ecc) [0x7fd48da4eecc]
# 4: /usr/lib/asterisk/modules/chan_dongle.so(+0x186a5) [0x7fd48da526a5]
# 5: /usr/lib/asterisk/modules/chan_dongle.so(+0xd1b0) [0x7fd48da471b0]
# 6: /usr/lib/asterisk/modules/chan_dongle.so(+0xdbd2) [0x7fd48da47bd2]
# 7: /usr/lib/asterisk/modules/chan_dongle.so(+0x12bbb) [0x7fd48da4cbbb]
# 8: /usr/sbin/asterisk(+0x18849f) [0x55e07cb7a49f]
# 9: /lib/x86_64-linux-gnu/libpthread.so.0(+0x7fa3) [0x7fd491914fa3]
#10: /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7fd49166f4cf]
```

532910, 2019-11-19:
Looks like it's due to `�` symbol at the end of Provider Name from  #39

closed 2019-11-20 by wdoekes

532910, 2019-11-20:
Just an acknowledgment that master works fine. Thank you, Walter!

BTW, do you know, why github shows "Latest commit 8b1827c on Jan 21" on the top of https://github.com/wdoekes/asterisk-chan-dongle, while recent commits are shown on the the commits page https://github.com/wdoekes/asterisk-chan-dongle/commits/master ?

wdoekes, 2019-11-20:
I do.

It's because I cherry-picked an old commit by garronej:
```
$ git show --format=fuller 8b1827c53 | grep -E 'Author|Commit'
Author:     Garrone Joseph <joseph.garrone@ensimag.grenoble-inp.fr>
AuthorDate: Mon Jan 21 10:41:12 2019 +0100
Commit:     Walter Doekes <walter+github@wjd.nu>
CommitDate: Wed Nov 20 09:49:30 2019 +0100
```

532910, 2019-11-20:
Is there a reason for this?
I always thought it shows the last activity, now I will not think so anymore. But in any case, this is confusing.

wdoekes, 2019-11-20:
I'm sorry. That's something you have to take up with GitHub. The text "Latest commit 8b1827c on Jan 21" is simply wrong. It should be "Latest commit 8b1827c on Nov 20", as that is the date the commit was done.

Note that this problem goes away when a new commit (that isn't an ancient cherry-pick) is added.

532910, 2019-11-20:
Pressing the commit link https://github.com/wdoekes/asterisk-chan-dongle/commit/8b1827c536b0a702608bb4c2f207e54d6c48b8f0 also shows "Jan 21".
Do you insist this is a github bug?

532910, 2019-11-20:
Looks like it shows the date commit was added (PR?) (AuthorDate: Mon Jan 21), but not approved by you (CommitDate: Wed Nov 20).