wdoekes/asterisk-chan-dongle #70: How to send SMS to shortcode ??

rgchurion, 2019-11-07:
Hi, how can i send SMS to shortcode for example 3535 ??, i try and get ERROR[5285]:

dongle sms dongle0 3535 registro

[dongle0] SMS queued for send with id 0x7f35340520e0

[Nov  7 18:48:25] ERROR[5285]: at_response.c:500 at_response_error: [dongle0] Error sending SMS message 0x7f35340520e0

I have installed this version of dongle in asterisk 16.

wdoekes, 2019-11-20:
Does this help:

https://github.com/miopa/asterisk-chan-dongle/commit/3a85d35c6689da1489e448944e089e7f390fde35

?

wdoekes, 2020-05-03:
I would indeed guess that:
```
#define NUMBER_TYPE_NETWORKSHORT	(TP_A_EXT_NOEXT | TP_A_TON_NETSPECIFIC | TP_A_NPI_PRIVATENUM) /* 0xB9 */
```
and
```
			dst_toa=NUMBER_TYPE_NETWORKSHORT; //0xB9
```
help.

But you'd have to test.

Closing for lack of feedback.

closed 2020-05-03 by wdoekes