wdoekes/asterisk-chan-dongle #73: how could I include this fork/project on a raspbx distribution ?

dandumit, 2019-12-28:
Hello,
Your project it's most active.
How could I include this in a raspbx distribution ?
I don't have access to so old huawei modems and I need to use this funtionality.
Thank you,
Daniel

ger-pakon, 2019-12-28:
Hey, it's not my project, but you can help to support newer huawei modems by testing and release information. Which modem do you want to use?

Cheers, 
Pakon

dandumit, 2019-12-28:
I have tested and seems to work (with some warnings) k3772 . this is not listed as compatible but working for me.  I have another one e362 that doesn't work.
I would be glad to help. hopefully with some added value.
Still my question stands : how I can include last version on my raspbx ?
Daniel

ger-pakon, 2019-12-28:
Hey Daniel,

I don't know the raspibox but I compiled it for myself on Rpi3. The output is one file for the Asterisk module directory. 

Cheers, 
Pakon

dandumit, 2019-12-28:
HEllo Pakon,
Is there any tutorial how to compile it ? I have installed on RPI 4. I think that should not be so different. 

Could you please share compilation procedure  ? 

In what new version of modems are you interested ?
Thank you,
Daniel

wdoekes, 2020-05-03:
Not sure what the issue is here. Closing. Please reopen if there is an actual issue that needs fixing.

closed 2020-05-03 by wdoekes