wdoekes/asterisk-chan-dongle #74: Trying to build automake: error: no 'Makefile.am'

dandumit, 2020-02-02:
HEllo All,
I am trying to build but I am getting this error :

automake: error: no 'Makefile.am' found for any configure output

What I am missing ?
Thank you,
DAniel

wdoekes, 2020-02-03:
Did you follow these steps?

https://github.com/wdoekes/asterisk-chan-dongle#buildin

ruijorgesilva1, 2020-02-25:
My problem is the same!

I've followed https://github.com/wdoekes/asterisk-chan-dongle#building and it hangs on

gcc -g -O2 -O6 -Wno-old-style-declaration -I. -std=gnu99 -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT smsdb.o -MF .smsdb.o.d -MP  -o smsdb.o -c smsdb.c
smsdb.c:26:21: fatal error: sqlite3.h: No such file or directory
 #include <sqlite3.h>

commit a4a7a345 (2020-05-03): Add configure.ac checks for sqlite3

Closes #74, reported by @ruijorgesilva1.
closed 2020-05-03 by wdoekes