wdoekes/asterisk-chan-dongle #78: smsdb.c error-message

hhornung, 2020-02-25:
with the newest release (2/24/20) I receive the following error-message at startup:
[2020-02-25 18:55:11] WARNING[10078] smsdb.c: Couldn't prepare statement 'INSERT INTO outgoing (key, refid) VALUES (?, 0) ON CONFLICT(key) DO UPDATE SET refid = CASE WHEN refid < 65535 THEN refid + 1 ELSE 0 END': near "ON": syntax error

How can I solve this?

best regards

ruijorgesilva1, 2020-02-25:
I have similar problem
When doing Make
gcc -g -O2 -O6 -Wno-old-style-declaration -I. -std=gnu99 -DAST_MODULE_SELF_SYM=__internal_chan_dongle_self -D_GNU_SOURCE -I/usr/include -I/usr/include -DHAVE_CONFIG_H  -fvisibility=hidden -fPIC -Wall -Wextra -MD -MT smsdb.o -MF .smsdb.o.d -MP  -o smsdb.o -c smsdb.c
smsdb.c:26:21: fatal error: sqlite3.h: No such file or directory
 #include <sqlite3.h>
                     ^
compilation terminated.
make: *** [smsdb.o] Error 1

hhornung, 2020-02-25:
@ruijorgesilva1: you have to install sqlite3 first: sudo apt-get install libsqlite3-dev

ruijorgesilva1, 2020-02-25:
@hhornung Thank you!
Everything went well!

Best Regards

magcks, 2020-02-27:
@hhornung Please check the current version form #77

hhornung, 2020-02-27:
still the same:
[2020-02-27 09:45:22] WARNING[6721]: smsdb.c:77 init_stmt: Couldn't prepare statement 'INSERT OR REPLACE INTO incoming (key, seqorder, expiration, message) VALUES (?, ?, datetime(julianday(CURRENT_TIMESTAMP) + ? / 86400.0), ?)': table incoming has no column named expiration
 [||] Got Response for user's command:'OK'

magcks, 2020-02-27:
Actually not the same. Do rm /var/lib/asterisk/smsdb.sqlite and restart asterisk.

hhornung, 2020-02-27:
yes - that's right. The message is gone - tx :)
raspbx*CLI> module load chan_dongle
Loaded chan_dongle
[2020-02-27 17:41:58] NOTICE[25152]: chan_dongle.c:1604 reload_config: [|] Loaded device
[2020-02-27 17:41:58] NOTICE[25152]: chan_dongle.c:1604 reload_config: [||] Loaded device
[2020-02-27 17:41:58] NOTICE[25152]: chan_dongle.c:1604 reload_config: [|||] Loaded device
[2020-02-27 17:41:58] NOTICE[25152]: chan_dongle.c:1604 reload_config: [||||] Loaded device
[2020-02-27 17:41:58] NOTICE[25152]: chan_dongle.c:1604 reload_config: [|||||] Loaded device
  == Registered channel type 'Dongle' (Huawei 3G Dongle Channel Driver)
    -- [|] Trying to connect on /dev/ttyUSB11...
    -- [|] Dongle has connected, initializing...
    -- [||] Trying to connect on /dev/ttyUSB14...
    -- [||] Dongle has connected, initializing...
    -- [|||] Trying to connect on /dev/ttyUSB8...
    -- [|||] Dongle has connected, initializing...
    -- [||||] Trying to connect on /dev/ttyUSB2...
    -- [||||] Dongle has connected, initializing...
    -- [|||||] Trying to connect on /dev/ttyUSB5...
    -- [|||||] Dongle has connected, initializing...
 [||] Got Response for user's command:'OK'
[2020-02-27 17:41:58] NOTICE[25155]: at_response.c:1794 at_response: [||] Got Response for user's command:'OK'
 [|] Got Response for user's command:'OK'
[2020-02-27 17:41:58] NOTICE[25154]: at_response.c:1794 at_response: [|] Got Response for user's command:'OK'
 [|||] Got Response for user's command:'OK'
[2020-02-27 17:41:58] NOTICE[25156]: at_response.c:1794 at_response: [|||] Got Response for user's command:'OK'
 [|||||] Got Response for user's command:'OK'
[2020-02-27 17:41:58] NOTICE[25158]: at_response.c:1794 at_response: [|||||] Got Response for user's command:'OK'
 [||||] Got Response for user's command:'OK'
[2020-02-27 17:41:58] NOTICE[25157]: at_response.c:1794 at_response: [||||] Got Response for user's command:'OK'
  == Registered application 'DongleStatus'
  == Registered application 'DongleSendSMS'
  == Registered application 'DongleSendUSSD'
  == Manager registered action DongleShowDevices
  == Manager registered action DongleSendUSSD
  == Manager registered action DongleSendSMS
  == Manager registered action DongleSetCCWA
  == Manager registered action DongleReset
  == Manager registered action DongleRestart
  == Manager registered action DongleStop
  == Manager registered action DongleStart
  == Manager registered action DongleRemove
  == Manager registered action DongleReload
 Loaded chan_dongle.so => (Huawei 3G Dongle Channel Driver)
    -- [||||] Dongle initialized and ready
    -- [|||||] Dongle initialized and ready
    -- [|] Dongle initialized and ready
    -- [|||] Dongle initialized and ready
    -- [||] Dongle initialized and ready

hhornung, 2020-02-27:
@magcks  But your latest build is crashing on unload (module unload chan_dongle) asterisk-Version is 13.31.0 on a raspberry pi 4 B

magcks, 2020-02-28:
I will check that, thank you.

hhornung, 2020-02-28:
The modem is a K3520 and sorry I have not checked all this stuff yet (calls are working).

magcks, 2020-02-28:
Can you please check the current version? If possible, can you also check the other stuff, please? It would be great so know if it works on other modems.

magcks, 2020-05-15:
This should be fixed now. The smsdb doesn't use any threads anymore now.