wdoekes/asterisk-chan-dongle #81: One way audio on FreePBX 15

crazyquark, 2020-03-22:
Hi,

I used previously chan_dongle with RasPBX but since then I migrated to another setup with Docker.
I've been trying to debug an issue I've had from the start:
- When a call comes from GSM, there is no audio on either end
- When a call is placed from PBX to GSM, only the PBX and receives the remote audio, the other end gets only silence...
Any clue about how to debug this would be great!

crazyquark, 2020-03-22:
The same happens on the latest RasPBX distro. It also uses kernel 4.19.

crazyquark, 2020-03-23:
Initially I thought it might be related to Asterisk version so I downgraded from 16 to 13 w/ FreePBX 15 but the problem persists.
Wierdly, on RaspPBX Stretch, kernel 4.14 w/ FreePBX 14 and Asterisk 13, it works fine.
So something broke... but I am not sure what or where.

vogelfreiheit, 2021-12-17:
Did you ever fix this with a recent distro and version of asterisk?