wdoekes/asterisk-chan-dongle #84: Errors with USSD, SMS

veerendra-nath, 2020-05-15:
**This is error When USSD:**

`[dongle0] USSD queued for send with id 0x73e00e98
[2020-05-15 11:00:28] ERROR[21487]: at_response.c:515 at_response_error: [dongle0] Error sending USSD 0x73e00e98
raspbx*CLI> dongle ussd dongle0 *99#
[dongle0] USSD queued for send with id 0x73e037a0
[2020-05-15 11:00:31] ERROR[21487]: at_response.c:515 at_response_error: [dongle0] Error sending USSD 0x73e037a0
raspbx*CLI> dongle ussd dongle0 *121#
[dongle0] USSD queued for send with id 0x73e03758
`

**When sending SMS without country code (Local number India) error , it can't able to send , before it can able send (no problem with operator ) , with country code it can able to send ;**

`raspbx*CLI> dongle sms  dongle0 9550148998 hello   
[dongle0] SMS queued for send with id 0x73e01238
[2020-05-15 11:04:21] ERROR[21487]: at_response.c:496 at_response_error: [dongle0] Error sending SMS message 0x73e01238
raspbx*CLI> dongle sms  dongle0 +919550148998 hello
[dongle0] SMS queued for send with id 0x73e03850
[2020-05-15 11:04:29] NOTICE[21487]: at_response.c:258 at_response_ok: [dongle0] Successfully sent SMS message 0x73e03850
[2020-05-15 11:04:30] WARNING[4051][C-00000051]: chan_sip.c:23255 func_header_read: This function can only be used on SIP channels.
[2020-05-15 11:04:30] WARNING[4051][C-00000051]: Ext. s:3 @ from-trunk: Friendly Scanner from 
[2020-05-15 11:04:32] WARNING[4051][C-00000051]: channel.c:1124 __ast_queue_frame: Exceptionally long voice queue length queuing to Local/ussd@from-trunk-dongle-0000001c;2
Incoming SMS from +919550148998 hello
[2020-05-15 11:04:33] WARNING[4051][C-00000051]: channel.c:1124 __ast_queue_frame: Exceptionally long voice queue length queuing to Local/ussd@from-trunk-dongle-0000001c;2
`

magcks, 2020-05-15:
Did you try to put a zero in front of your national number (or whatevery indian operators use as local exit prefix; in europe it's a zero...)? Beforehands, everything was threated as NUMBER_TYPE_INTERNATIONAL, now numbers without a + are threated as NUMBER_TYPE_UNKNOWN.

veerendra-nath, 2020-05-15:
with latest merge , there is no problem with out country code 
but i can't able to send message to operator service numbers 

`raspbx*CLI> dongle sms dongle0 1900 hello     
[dongle0] SMS queued for send
    -- [dongle0] Error payload: UNKNOWN
    -- [dongle0] Error sending SMS message 0x73f055e8
[2020-05-15 16:30:24] ERROR[12321]: at_response.c:516 at_response_error: [dongle0] Error sending SMS message 0x73f055e8 SMSTEXT
`

magcks, 2020-05-15:
Did this work before? Sounds strange, as @miopa introduced the NUMBER_TYPE_NETWORKSHORT case. Could you please try to replace NUMBER_TYPE_NETWORKSHORT on line 568 of pdu.c with NUMBER_TYPE_INTERNATIONAL?

veerendra-nath, 2020-05-15:
@magcks  no , i didn't tested before , i recently stating using FreePBX

veerendra-nath, 2020-05-15:
@magcks  i replaced NUMBER_TYPE_NETWORKSHORT with  NUMBER_TYPE_INTERNATIONAL
same problem 
`NUMBER_TYPE_INTERNATIONAL`

wdoekes, 2020-05-15:
> (or whatevery indian operators use as local exit prefix; in europe it's a zero...)

Actually. I don't think that applies to all countries. It does to NL and DE. But AFAIK, Greece, Spain, Portugal, Luxembourgh and a bunch of other countries don't use a 0-prefix when dialing nationally.

See for instance https://en.wikipedia.org/wiki/Telephone_numbers_in_Greece#Overview
> Finally, in 2002, the leading '0' was changed to a '2' (for geographic numbers)

wdoekes, 2020-05-15:
@veerendra-nath : in your first comment you said "before it can able send (no problem with operator )".

That leads us to believe that something was changed in chan_dongle recently. Is there a specific commit that changed things after which things stopped working for you?

Please use `git bisect` to figure out which commit breaks things for you.

veerendra-nath, 2020-05-15:
> @veerendra-nath : in your first comment you said "before it can able send (no problem with operator )".
>
> That leads us to believe that something was changed in chan_dongle recently. Is there a [...]

sorry for my bad , that one i used old chan_dongle lib comes with RaspPBX distribution

miopa, 2020-05-17:
What's the status with this?

The patch I made was tested with network short, local and international numbers on multiple operators and countries in North America, Europe and East Africa. Without the patch, SMS sending was failing on some operators with short and local numbers.

magcks, 2020-05-18:
@miopa I think @veerendra-nath used the old Version from that raspberry pi repo. Your patch is working for me at least for national numbers. I didn't try operator numbers as we don't really use them in germany.

archekb, 2021-01-29:
Hi! I use commit 2bd3fc2cc31562826c8341c5584fe26def9ca808 of https://github.com/wdoekes/asterisk-chan-dongle, and try send sms to short code:
> asterisk*CLI> dongle sms gsm912xxxxxxx 3620 Z4088924*78995
> [gsm912xxxxxxx] SMS queued for send
> [Jan 29 07:22:11] ERROR[82]: at_response.c:287 log_cmd_response_error: [gsm912xxxxxxx] Error sending SMS message 0x7f5229f7ec70 SMSTEXT

In main console:
> asterisk_1  | [Jan 29 07:22:11]     -- [gsm912xxxxxxx] Error payload: UNKNOWN
> asterisk_1  | [Jan 29 07:22:11]     -- [gsm912xxxxxxx] Error sending SMS message 0x7f5229f7ec70
> asterisk_1  | [Jan 29 07:22:11] ERROR[82]: at_response.c:287 log_cmd_response_error: [gsm912xxxxxxx] Error sending SMS message 0x7f5229f7ec70 SMSTEXT
> asterisk_1  | [Jan 29 07:22:11]     -- Executing [report@in_gsm912xxxxxxx:1] Hangup("Local/report@in_gsm912xxxxxxx-00000000;1", "") in new stack
> asterisk_1  | [Jan 29 07:22:11]   == Spawn extension (in_gsm912xxxxxxx, report, 1) exited non-zero on 'Local/report@in_gsm912xxxxxxx-00000000;1'
> asterisk_1  | [Jan 29 07:22:11]     -- Executing [h@in_gsm912xxxxxxx:1] Hangup("Local/report@in_gsm912xxxxxxx-00000000;1", "") in new stack
> asterisk_1  | [Jan 29 07:22:11]   == Spawn extension (in_gsm912xxxxxxx, h, 1) exited non-zero on 'Local/report@in_gsm912xxxxxxx-00000000;1'