wdoekes/asterisk-chan-dongle #85: after recent merger chan_dongle unable to load

veerendra-nath, 2020-05-15:
This is the log message 

[2020-05-15 12:17:12] ERROR[1340]: loader.c:2396 load_modules: Error loading module 'chan_dongle.so': /usr/lib/asterisk/modules/chan_dongle.so: undefined symbol: at_response_cdsi

wdoekes, 2020-05-15:
@magcks: sounds familiar?

veerendra-nath, 2020-05-15:
commenting 'at_response_cdsi' function call in switch statement , its working 
but incoming sms are not working, 
`[2020-05-15 14:10:52] WARNING[8905]: at_response.c:1226 at_response_cmti: [dongle0] Error parsing incoming sms message alert '+CDSI: "SM",21', ignoring
`
this error i am getting

veerendra-nath, 2020-05-15:
I compiled with latest pull request 71e3206 , the module is loaded 
now i am not getting incoming SMS's 
core restart now, after this i am getting the all pending  SMS's  at instance , after core restarts same problem repeat
in simple i have to restart the core when i want see my latest SMS's

veerendra-nath, 2020-05-15:
`core restart now` or `module reload chan_dongle.so` , i have to do this to get the pending incoming  SMS's 

> I compiled with latest pull request [71e3206](https://github.com/wdoekes/asterisk-chan-dongle/commit/71e3206122a36f45f097a712a3c8cb611db365c4) , the module is loaded
> now i am not getting incoming [...]

wdoekes, 2020-05-15:
Also here: please find out which commit broke things for you (`git bisect`)

magcks, 2020-05-18:
Probably this is the same problem as #87

wdoekes, 2020-05-28:
(Please try master.)

wdoekes, 2020-06-25:
Fixed? Probably?

closed 2020-06-25 by wdoekes