wdoekes/asterisk-chan-dongle #87: State shows as SMS instead of Free

turbozapekanka, 2020-05-16:
Today's build
[image]

wdoekes, 2020-05-16:
@turbozapekanka : can you pinpoint which version breaks things for you?

`git bisect`

magcks, 2020-05-16:
I would suggest to test a4a7a345e582ee3b15ca3f6142409e4ec2d5c7ba. There were some changes in the fetching logic in #79 (ba5e7215271283d555fd60083ce0d5d71dafe73e)

turbozapekanka, 2020-05-16:
It was on https://github.com/wdoekes/asterisk-chan-dongle/commit/7bdef5e6627eafc7d20a9c893f8bd1b781aec8b7

Will do more tests

ljk4160, 2020-05-18:
I have the same problem,can send out and State displays SMS,
When receiving SMS, the log displays SMS is full
My device is E180, asterisk version is 13.33
if asterisk v16.9.0 not shown SMS is full,but Can not receive any SMS message
I tried to reinstall the OS, but I also encountered the above problems

turbozapekanka, 2020-05-18:
I'm using freepbx distro (I beleive it is centos 7 based) with asterisk 13.32.0. Not sure if it makes any difference. the kernel is quite old tho. 3.10 or something

ljk4160, 2020-05-18:
> I would suggest to test [a4a7a34](https://github.com/wdoekes/asterisk-chan-dongle/commit/a4a7a345e582ee3b15ca3f6142409e4ec2d5c7ba). There were some changes in the fetching logic in #79 ([ba5e721](https://github.com/wdoekes/asterisk-chan-dongle/commit/ba5e7215271283d555fd60083ce0d5d71dafe73e))

use a4a7a34 is ok

magcks, 2020-05-18:
Thank you. Could you please also test https://github.com/magcks/asterisk-chan-dongle/commit/868bbd41fbf2911cdfba483e50c8e99a79d84258

ljk4160, 2020-05-19:
> Thank you. Could you please also test [magcks@868bbd4](https://github.com/magcks/asterisk-chan-dongle/commit/868bbd41fbf2911cdfba483e50c8e99a79d84258)

WARNING[8392]: at_response.c:1280 at_response_cmgr: [dongle0] Error parsing in                coming message: Cannot parse UCS-2

miopa, 2020-05-28:
> Thank you. Could you please also test [magcks@868bbd4](https://github.com/magcks/asterisk-chan-dongle/commit/868bbd41fbf2911cdfba483e50c8e99a79d84258)

This version seems to work for me (on Asterisk 13.31.0)

magcks, 2020-05-28:
Okay, in this case there probably issues with #79.

hselasky, 2020-05-28:
I'll have a look later today syncing up with the latest code. The SMS state is not critial. From what I understand it is just a read-only status flag.

hselasky, 2020-05-28:
Can you try this one:
https://github.com/wdoekes/asterisk-chan-dongle/pull/90

hselasky, 2020-05-28:
I'm seeing the same issue with latest version (not mine). Must be a merge issues.

hselasky, 2020-05-28:
OK,
This revision works for me:
413cad90774f6ef547abf239e6101129bbe4d20d

This revision doesn't:
71e3206122a36f45f097a712a3c8cb611db365c4

Problem is related to missing response to AT+CMGR=<n> which fetch SMS'es.

hselasky, 2020-05-28:
Hi,
This patch fixes it. It's a regression after the PDU parsing restructure. (No my fault, but now it is fixed :-)

https://github.com/wdoekes/asterisk-chan-dongle/commit/a91275d119fdb5f4e3069692a611ca257c47da33

--HPS

hselasky, 2020-05-28:
Please retest #90 !

Thank you!

wdoekes, 2020-05-28:
(Or master, as I merged just now.)

magcks, 2020-05-29:
Thank you very much @hselasky

turbozapekanka, 2020-06-08:
Yes I confirm the state now shows Free but there are some error messages https://github.com/wdoekes/asterisk-chan-dongle/issues/97

turbozapekanka, 2020-06-17:
Confirming PR https://github.com/wdoekes/asterisk-chan-dongle/pull/90 fixes this

closed 2020-06-17 by turbozapekanka