wdoekes/asterisk-chan-dongle #88: master is not compiling against asterisk 1.8

miopa, 2020-05-28:
last revision that works is fe3b185f35b8096763ca782b79479976926d9e78

Compiling on CentOS 6 with gcc 4.4.7

```
at_command.o: In function `ast_tvdiff_sec':
/usr/include/asterisk/time.h:45: multiple definition of `ast_tvdiff_sec'
app.o:/usr/include/asterisk/time.h:45: first defined here
```

wdoekes, 2020-05-28:
Hm. I'm afraid that might be intentional. @magcks added new code after that and he didn't plan on making everything backwards compatible. (Which I fully understand.)

There are a few things we can do:
- tag fe3b185 as the last version that works with 1.8 and document that (easiest)
- add fixes to make master work with 1.8 again (unknown how much effort/time that takes)

I'm kind of waiting for some conclusive fixes for #85 and #87 before I feel like touching the repo though.

In all cases: thanks for reporting :+1:  This info will be useful to others

magcks, 2020-05-28:
1.8 is officially not supported anymore by asterisk.

wdoekes, 2020-05-28:
> 1.8 is officially not supported anymore by asterisk.

Yes, but doesn't _mandate_ that we drop support for it.

miopa, 2020-05-29:
Another argument is that because of the cheap Chinese VoIP boxes, the number of running Asterisk instances with versions 1.6/1.8 is an order of magnitude bigger than all newer versions combined

wdoekes, 2020-06-25:
So, I assume #85 and #87 are fixed now.

It looks like you need this:
```
Subject: use fix from inline_api.h from Asterisk 11

--- asterisk-1.8/include/asterisk/inline_api.h	2016-10-02 13:39:07.000000000 +0200
+++ asterisk-1.8/include/asterisk/inline_api.h	2016-08-29 11:50:08.000000000 +0200
@@ -46,8 +46,12 @@
 #if !defined(LOW_MEMORY)
 
 #if !defined(AST_API_MODULE)
+#if defined(__clang__) || defined(__GNUC_STDC_INLINE__)
+#define AST_INLINE_API(hdr, body) static hdr; static inline hdr body
+#else /* if defined(__clang__) */
 #define AST_INLINE_API(hdr, body) hdr; extern inline hdr body
-#else
+#endif
+#else /* if !defined(AST_API_MODULE) */
 #define AST_INLINE_API(hdr, body) hdr; hdr body
 #endif
 
```

miopa, 2020-06-26:
Yes, builds and works! Thanks

closed 2020-06-26 by miopa