wdoekes/asterisk-chan-dongle #89: chan_dongle.so not compiled with the same compile-time options

Yannik, 2020-05-28:
Hi,
I built chan_dongle like this:

```
./bootstrap
./configure --with-astversion=16.2
make
make install
```

The platform I'm on is debian 10 (x64).
The installed asterisk version is:
`ii  asterisk                          1:16.2.1~dfsg-1+deb10u1                                        amd64`

Unfortunately, asterisk fails to load the module:
```
[2020-05-28 11:08:46] ERROR[7621]: loader.c:170 module_load_error: Module 'chan_dongle.so' was not compiled with the same compile-time options as this version of Asterisk.
[2020-05-28 11:08:46] ERROR[7621]: loader.c:170 module_load_error: Module 'chan_dongle.so' will not be initialized as it may cause instability.
[2020-05-28 11:08:46] ERROR[7621]: loader.c:170 module_load_error: Module 'chan_dongle' could not be loaded.
```

How can I fix this?

Yannik, 2020-05-28:
Fixed by installing the `asterisk-dev` package.

closed 2020-05-28 by Yannik

wdoekes, 2020-05-28:
Thanks for following up with your fix :+1:

Yannik, 2020-05-28:
@wdoekes Sure. I think mentioning this requirement in the README would be a great idea.. :-)

Yannik, 2020-05-28:
@wdoekes Or otherwise the configure script could throw an error if the required header files are not found.

lpcs007, 2023-01-09:
I'm trying to install the module on Asterisk 18.15.0 and I'm having the same error.
I tried 18.15.0 and 18.15.
the asterisk-dev package is already installed

I'm using VitalPBX 4