wdoekes/asterisk-chan-dongle #91: Sending and receiving SMS do not working anymore from dec.2019

easyyu, 2020-05-28:
Hello..
I upgrade the asterisk from 13.29.2 to 13.33 and even compile again the chan_dongle today.
From this moment I can not send neither receive sms. The problem is definetly in chan_dongle couse if I put back in `/usr/lib/asterisk/modules` file **chan_donle.so** from 7.dec.2019 everything working OK.
Interesting is that when I send sms I apsolutly do not see that chan_dongle triger the asterisk.
As my dialplan is not any more ok. I do not think that there is any purpose to copy here my dialplan couse there is not problem with him, its working with older (dec.2019) comiled chan_dongle.
Interesting is that when I receive the sms the asterisk CLI just jump out. Asterisk from himself is OK.
Calls are ok (incoming and outgoing).
Compiling pass without any problem, I tried twice and watched the proccess.
If u need some aditional log or any kind of info I'm with will to post here.

My config is: Armbian Debian 10.2 on orangepi
Huawei E1550 dongle modem.

Any advice?

turbozapekanka, 2020-05-28:
probably related https://github.com/wdoekes/asterisk-chan-dongle/issues/87

easyyu, 2020-05-28:
thx for answer..
Can u help me how to git clone that changes, because I see that he just finished and tested the code one hour ago.

wdoekes, 2020-05-28:
Please try master.

easyyu, 2020-06-24:
Sorry for late answer..coudn;t test before.
Yes, the receiving of messages are now OK, but sending NOW do not working!
I have now the error messages:

> - Executing [0655305839@user-SMS:14] DongleSendSMS("Message/ast_msg_queue", "dongle0,+38xxxxxxxx,"Helo",1440,yes") in new stack
> **[Jun 24 15:43:06] ERROR[29451][C-00000000]: app.c:111 app_send_sms_exec: NULL payload for message -- SMS will not be sent**
>   == Spawn extension (user-SMS, 0655305839, 14) exited non-zero on 'Message/ast_msg_queue'

What can be the problem now?
My dialplan is OK, and its working with release of dec.2019
Thx

hardillb, 2020-09-02:
I'm going to guess this is because your dial plan looks something like this:
```
exten => _.,n,DongleSendSMS(dongle0,${EXTEN},${MESSAGE(body)},1440,no)
```

It looks like it now requires a `payload` value after the yes/no for the delivery report. Looking at the code the payload should only be used if there is a delivery report, but a recent change makes the payload mandatory even if it's set to no. Try adding `"foo"`  on the end of the DongleSendSMS line like this:

```
exten => _.,n,DongleSendSMS(dongle0,${EXTEN},${MESSAGE(body)},1440,no,"foo")
```

easyyu, 2020-09-02:
Yes I realised..just forgot to announce here.
Thx anyway!

closed 2020-09-02 by easyyu