wdoekes/asterisk-chan-dongle #93: Error parsing incoming message @ 328b2b7

ljk4160, 2020-05-30:
WARNING[24354]: at_response.c:1311 at_response_cmgr: [dongle0] Error parsing incoming message: Cannot parse UCS-2

OS:freepbx.sangoma(centos?)
asterisk version:16.6.2
3g dongle:HUAWEI E180

wdoekes, 2020-05-30:
So, did this problem start in 328b2b7 ?

Which is the last version that works for you?

Help us help you.

ljk4160, 2020-05-30:
I now use a4a7a34 is ok, I've tried 868bbd4 later versions of this error

ljk4160, 2020-05-30:
I thought the latest version was fixed, so I tested the latest version again, and found that the problem still exists

wdoekes, 2020-05-30:
And the versions between those two versions? `git bisect` is your friend.

Although 868bbd4 indeed does rework a lot -- including character conversions -- so it is likely related.

It probably also helps if you can add some debug messages -- including the PDU so the failing message can be looked at and the problem can be reproduced.

qiusj, 2020-07-17:
WARNING[3934]: at_response.c:1267 at_response_cmgr: [dongle0] Error parsing incoming message: Cannot parse UCS-2
OS:freepbx.sangoma
asterisk: version:16.6.2
3g dongle:HUAWEI E150
chan_dongle: the last version

wdoekes, 2021-10-01:
Probably fixed by now.

closed 2021-10-01 by wdoekes