wdoekes/asterisk-chan-dongle #94: Can't send SMS

ger-pakon, 2020-05-30:
Hey,

I use `7bdef5e` and can't send a sms anymore. In this case I get following error:

server*CLI> dongle sms dongle1 <german_mobile_number> test
[dongle1] SMSDB error
[May 30 22:20:49] WARNING[4911]: smsdb.c:420 smsdb_get_refid: Couldn't bind key to stmt: not an error

I will edit this post, when I find the commit witch it breakes...

Thank you,
pakon

wdoekes, 2020-05-31:
Did you try latest? 7bdef5e is not latest.

ger-pakon, 2020-05-31:
I just can checkout this version. 

```
commit 328b2b7ded3b3c59c2e6c7e7afca6b0ac6f146ce
Merge: 7bdef5e 5110695
Author: Walter Doekes <walter+github@wjd.nu>
Date:   Thu May 28 21:57:10 2020 +0200

    Merge pull request #90 from hselasky/master

    Fixes for regression issues and some improvements.

    Thanks Hans Petter Selasky!
```

Otherwise I don't know how to get the latest. 

Thank you,
pakon

ger-pakon, 2020-06-21:
I found the problem. As described in https://github.com/wdoekes/asterisk-chan-dongle/issues/78#issuecomment-592056692 I just had to delete `/var/lib/asterisk/smsdb.sqlite3` and restart Asterisk. 

Thank you,
pakon

closed 2020-06-21 by ger-pakon

hanyuanzhong, 2024-07-25:
> 我找到了问题所在。如[#78 (评论)](https://github.com/wdoekes/asterisk-chan-dongle/issues/78#issuecomment-592056692)中所述，我只需删除`/var/lib/asterisk/smsdb.sqlite3`并重新启动 Asterisk 即可。
> 
> 谢谢你， pakon

@wdoekes hi
I also met this problem, I use a raspberry sent 4 b + huawei k3765 encryption dog, when I delete the/var/lib/asterisk/SMSDB sqlite3, and restart the asterisk after successfully send a text message, but my raspberries pie after the restart, They will report the error, I still need to manually delete the/var/lib/asterisk/SMSDB sqlite3, and restart the asterisk, this is what causes, what other solutions, please help me
[image]