wdoekes/asterisk-chan-dongle #95: PIN suppport for SIM cards

p-velasco, 2020-06-02:
Hello,

Is there any reason why PIN functionality is not supported? I have seen a "hackish" patch at https://github.com/pld-linux/asterisk-chan_dongle/blob/master/chan_dongle-pin.patch, and wonder why it hasn't been implemented yet.

So I think I'm missing something preventing it, and would like to know what before starting the route of modifying sources to reach that dead end (if any).

Maybe is due to chan_dongle querying the IMSI (which needs the card to be unlocked)?
If so, maybe the IMSI approach could be discarded and instead use the ICCID, which can be queried anytime via the AT^ICCID? command.

wdoekes, 2020-06-02:
I know of no reason why it should not be implemented.

PRs are welcome ;-)

p-velasco, 2020-06-03:
Ok, will try if I get some spare time. While I try to understand the code flow (any help with that?), maybe my PR for some code improvements can be accepted?