wdoekes/asterisk-chan-dongle #110: Couldn't communicate to E160G modem

grwlf, 2020-08-02:
Hi. I have E160G modem which seems to work correctly with minicom (`sudo minicom -D /dev/ttyUSB0`) and [reference C program](https://gist.github.com/grwlf/5e0cea0154c9ce4794489a2e1dc93b76). Namely, I could easily send AT commands like 'AT' or 'ATI'.

Unfortunately, asterisk-chan-dongle reports errors like the below

```
*CLI> dongle cmd dongle0 ATI
[dongle0] 'ATI' Device disconnected
*CLI>     -- [dongle0] Trying to connect on /dev/ttyUSB0...
[Aug  2 11:56:47] WARNING[29293]: chan_dongle.c:284 opentty: Open /dev/ttyUSB0 successfully
      dongle cmd dongle0 ATI
[Aug  2 11:56:53] WARNING[29293]: chan_dongle.c:284 opentty: Open /dev/ttyUSB1 successfully
    -- [dongle0] Dongle has connected, initializing...
[Aug  2 11:56:53] ERROR[29215]: at_queue.c:251 at_queue_run: [dongle0] Error write command 'USER'S' expected response 'OK' length 4, cancel
[dongle0] 'ATI' Queue error
*CLI> [Aug  2 11:56:53] ERROR[29503]: at_queue.c:251 at_queue_run: [dongle0] Error write command 'AT' expected response 'OK' length 3, cancel
[Aug  2 11:56:53] ERROR[29503]: chan_dongle.c:454 do_monitor_phone: [dongle0] Error adding initialization commands to queue
    -- [dongle0] Error initializing Dongle
    -- [dongle0] Dongle has disconnected
    -- [dongle0] Trying to connect on /dev/ttyUSB0...
```

Could you please suggest what could be wrong?

Note, that `Open /dev/ttyUSB1 successfully` message was added by me, it is the only change. I'm using latest asterisk-chan-dongle (git hash 328b2b7) and asterisk 15.

grwlf, 2020-08-26:
Got some new info for this case:

I disabled opening `audio_fd` in `chan_dongle.c:715` (wrote -1 to it and commented out sanity checks). After that device could pass AT-checks and receive some SMS (I got messages ` [dongle0] Got full SMS from Beeline@ ....`). What could it mean? Could an attempt to open audio port hang the modem entirely?

grwlf, 2020-08-29:
I just got another modem which seems to work better. For now I believe that the problem was in my particular hardware.

closed 2020-08-29 by grwlf

wdoekes, 2020-08-29:
Thanks for the feedback / self-close :)