wdoekes/asterisk-chan-dongle #111: Compile fail on ast<17 because of missing ast_log_ap

wdoekes, 2020-08-24:
4652c90431e51722ee027e1444480e43d420594f#commitcomment-41706256:

> If I read this commit correctly here asterisk/asterisk@e4cf513 then ast_log_ap does not exist until Asterisk 17. Module cannot be loaded with this code on Asterisk 13.

http://c-faq.com/varargs/handoff.html

Looks like we should sprintf and pass a single argument instead..

grwlf, 2020-08-26:
Same conclusion here. Please, fix!

wdoekes, 2020-08-27:
> Same conclusion here. I'm using Asterisk 15.7. Please, fix!

Wrong tab, you should be in the "Pull requests" tab :smiling_imp: 

As this is a trivial issue, I'm sure you can manage that :)

grwlf, 2020-08-27:
> > Same conclusion here. I'm using Asterisk 15.7. Please, fix!
> 
> Wrong tab, you should be in the "Pull requests" tab 

Umm, quick search through `include/logger.h` of asterisk15 doesn't reveal to me anything comparable to `ast_log_as`, so it looks like I'm in trouble here..

> 
> As this is a trivial issue, I'm sure you can manage that :)

My workaround is .. well .. I've managed to compile asterisk17

wdoekes, 2020-09-02:
Thanks @hardillb

closed 2020-09-02 by wdoekes