wdoekes/asterisk-chan-dongle #113: Definative statment on supported Asterisk versions

hardillb, 2020-09-01:
If I'm reading #111 correctly the current version of the code requires Asterisk v17.

Assuming this is confirmed I'll submit a pull request to update the README.md as the current one implies that 13.x is supported in the build instructions.

wdoekes, 2020-09-01:
> If I'm reading #111 correctly the current version of the code requires Asterisk v17.

Yes.

> Assuming this is confirmed I'll submit a pull request

That's a bit premature. The fact that 111 implies that 17 is the only working version does not make it intentional.

You could do a PR to fix 111 instead *hint* ;-)

wdoekes, 2020-09-01:
*(For some context: I don't use (and have never used) chan_dongle. I only ever compile the thing to see if it builds. I don't check if it works, as I don't even have a dongle. This means that I sometimes manage to merge a PR that breaks it in obvious ways to those that actually do use it, without me noticing it. In those cases: feedback and PRs are gladly welcomed!)*

hardillb, 2020-09-01:
OK, I'll have a look at fixing the underlying problem.

I have confirmed that if I comment out the `ast_log_ap()` lines in at_responce.c it all works with v13.14.

closed 2020-09-02 by wdoekes