wdoekes/asterisk-chan-dongle #117: Unexpected service closeing

gonsalez-ru, 2020-09-18:
After executed in dialplan DongleSendSMS application Asterisk shuting down:

 [messages]
 exten => _X.,1,NoOp("${EXTEN} ${MESSAGE(body)}")
 exten => _X.,n,DongleSendSMS(gsm1,${EXTEN},"${MESSAGE(body)}")
 exten => h,n,Hangup()

Asterisk Ready.
    -- Executing [+79161234567@messages:1] NoOp("Message/ast_msg_queue", ""+79161234567 456"") in new stack
    -- Executing [+79161234567@messages:2] DongleSendSMS("Message/ast_msg_queue", "gsm1,+79161234567,"456"") in new stack
gate*CLI>
Disconnected from Asterisk server
Asterisk cleanly ending (0).
Executing last minute cleanups
Asterisk ending (0). 

P.S.
CLI "dongle sms" command works fine

gonsalez-ru, 2020-09-19:
I need your advise. What diagnostic actions can i do to gather additional debug information?

wdoekes, 2020-09-19:
Start the asterisk daemon from `gdb`:
```
# gdb `which asterisk`
(gdb) run asterisk -c
```
And then do the dongle stuff.

Then when it crashes, do:
```
bt
bt full
```
That should show you some info.

gonsalez-ru, 2020-09-19:
Since i use precompiled for mips router chan_dongle module i'm limited in debugging toools. I didn't find GDB packet for my platform and I'm not familar with exotic cross compilation. :-(

wdoekes, 2020-09-19:
Perhaps dmesg tells you at least an address (`ip` instruction pointer). Which you might be able to look up through `addr2line`. You may need to do some hexadecimal math to get the right value.

gonsalez-ru, 2020-10-02:
Here is dmesg out:
do_page_fault(): sending SIGSEGV to asterisk for invalid read access from 00000000
epc = 76b34540 in libc-2.27.so[76aad000+17a000]
ra  = 756eb6f0 in chan_dongle.so[756e8000+3c000] 

There are two packets installed with libc:
opkg files libc | grep libc-2.27.so
/opt/lib/libc-2.27.so 

 opkg files gcc | grep libc-2.27.so
/opt/lib/gcc/mipsel-openwrt-linux-gnu/7.4.0/libc-2.27.so