wdoekes/asterisk-chan-dongle #121: E173 modem receives SMS only during initialization

grwlf, 2020-10-16:
Related to E173 Huawei modem. Dongle seems to communicate with it correctly. I could send SMS from Asterisk CLI. Unfortunately, receiving SMS works only during the initialization of dongle (either on Asterisk restart or after `dongle restart now dongle0`: I can’t see any reaction from the software when I send SMS from my phone to modem’s simcard number, if asterisk is already running.

Configuration (please ignore unusual format, they are built-in into NixOS expression):

* Asterisk version: 17
* dongle.conf
* extensions.conf

Some status commands:

*CLI> dongle show devices
ID           Group State      RSSI Mode Submode Provider Name  Model      Firmware          IMEI             IMSI             Number        
dongle0      0     Free       16   0    0       Beeline        E173       11.126.85.00.209  35561504xxxxxx  250997216xxxxx  Unknown       
*CLI> 
*CLI> dialplan show dongle-incoming
[ Context 'dongle-incoming' created by 'pbx_config' ]
  'sms' =>          1. Verbose(Incoming SMS)                      [extensions.conf:4]
                    2. Hangup()                                   [extensions.conf:5]

-= 1 extension (2 priorities) in 1 context. =-
*CLI>

grwlf, 2020-10-20:
Talking to myself: The issue is solved after executing
```
dongle cmd dongle0 AT^PORTSEL=1
```

Thanks, https://github.com/antirek/asterisk-chan-dongle/issues/147

miopa, 2020-11-11:
I have some E173 with the same firmware as you, they receive SMS without any issue on asterisk 1.8, 13 and 15 (at least with my fork https://github.com/miopa/asterisk-chan-dongle)

> Talking to myself: A workaround for this issue is
> 
> ```
> dongle cmd dongle0 AT^PORTSEL=1
> ```

I tried this on two E173 (ast 1.8), both modems got stuck on 'dialing' soon

rusxakep, 2020-11-13:
Same problem with:

E171       21.156.00.00.143
E171       21.157.00.01.143